Unity-Technologies/URDF-Importer #11: Invalid AABB a erros when running

Transzendental, 2020-11-28:
I tried importing the sawyer robot (https://github.com/RethinkRobotics/sawyer_robot). The importing process worked well, except some minor issues with rotation which where easy enough to fix and may have been caused by a faulty urdf file.
However when im running the application ill get a lot of "Invalid AABB a" and similar erros. Also unity crashes when switching to the scene tab.

Right now i have no idea how to troubleshoot. Perhaps you are able to assist? With the niryo_one robot everything worked fine.

Coloradete, 2020-12-02:
I had a similar problem with the Baxter robot. I fixed it giving layers to the arms and the parts with colliders and disabling the physics collisions between them under the project settings -> Physics tab. hope that helps

mpinol, 2021-01-05:
The URDF Importer has been updated so that it allows for a 'disable collision' tag in imported URDFs which I believe should help both of your cases. You can find a more detailed description here, https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md#disable-collision-support

The change should be included in URDF Importer version 0.0.5, which should hopefully be released sometime today or tomorrow.

@Transzendental , Can you please add the disable collision tag to your URDF, reimport the robot, and then let me know the result?

@Coloradete , would you also mind taking a look at our update to see if it would resolve your issue while also being simpler to implement?

Thank you!

mrpropellers, 2021-01-20:
Hey @Coloradete and @Transzendental I'm doing a backlog sweep and would like to resolve this issue. Did the recent changes to the URDF Importer resolve your issues?

Transzendental, 2021-01-20:
Ive switched to ROS# on a non beta unity version instead. Since i no longer have the project im unfortunately not able to retest this. Feel free to close this issue.

closed 2021-01-25 by mpinol

adverley, 2021-02-03:
For Baxter, there are collision penetrations inside each arm and between the arms and the torso mesh collider. I can confirm the 'disable collision' tag resolves this issue for the common Baxter URDF file by creating tags for all combinations of the lists (torso, left_arm_links) and (torso, right_arm_links).

WenqiangX, 2021-04-01:
For the Franka Panda arm, I found the 'disable collision' cannot fix the Invalid AABB errors.

1. the Franka panda arm urdf can be loaded successfully in Pybullet.
2. Unity version is 2020.2 and 2020.3, URDF importer 0.2.0, both have invalid AABB errors.
3. When I use URDF importer 0.0.5 in Unity 2020.3, the MessProcess cannot be found.
Can anyone help? Thanks in advance!

pogopo789, 2023-08-24:
I have the same error when pressing play and I found that in some URDF Inertial (Script) there are NaN in Using URDF Data. By unchecking the box Use URDF Data. The problem is fixed.