Unity-Technologies/URDF-Importer #24: Velocity controller

silasalves, 2020-12-22:
I successfully imported a Turtlebot 2 model using the URDF importer (repositories for [turtlebot](https://github.com/turtlebot/turtlebot/tree/melodic/turtlebot_description) and [kobuki](https://github.com/yujinrobot/kobuki_description) URDF), however I am having trouble to move it.

I am able to change the wheels' angles using:

```c#
public class turtlebot_main : MonoBehaviour
{
    public ROSConnection ros;
    public string topicName = "SourceDestination_input";
    public GameObject turtlebot;
    private ArticulationBody[] wheels;

    void Awake()
    {
        string wheel_r = "base_footprint/base_link/wheel_right_link";
        string wheel_l = "base_footprint/base_link/wheel_left_link";

        wheels = new ArticulationBody[2];        
        wheels[0] = turtlebot.transform.Find(wheel_r).GetComponent<ArticulationBody>();
        wheels[1] = turtlebot.transform.Find(wheel_l).GetComponent<ArticulationBody>();
    }

    public void move()
    {
        var a = wheels[0].xDrive;
        a.target = 10f;
        wheels[0].xDrive = a;

        var b = wheels[1].xDrive;
        b.target = 10f;
        wheels[1].xDrive = b;
    }
}
```

but that is not the behavior I need.

I noticed that Position Control is the only controller type available so far. Are there plans to add a Velocity Controller as well? If not, how could I build such controller?

Thanks a lot!

vidurvij-Unity, 2020-12-22:
Hi @silasalves ,
Thank you for trying out URDF-Importer. We are working on introducing velocity control as a standard controller type in URDF. But you can easily make a velocity controller yourself. You need to set the target velocity of the xDrive component of the Articulation Body you want to control. 
And also make sure you have set a non zero value for damping and forceLimit for the xDrive and set the stiffness to zero.

You can find more details on the APIs here : https://docs.unity3d.com/2020.1/Documentation/ScriptReference/ArticulationDrive.html
Also keep in mind that the targetVelocity is in degree per second. 
Please tell if this solution helps and feel free to reach out again.
Vidur Vij

silasalves, 2020-12-22:
Thanks, @vidurvij-Unity, setting the stiffness to zero did the job! The robot is now moving as expected.

closed 2020-12-22 by silasalves