Unity-Technologies/URDF-Importer #25: importing PR2 robot URDF causes crash

WenqiangX, 2020-12-30:
Unity version: 2020.2.1f1c1
URDF Importer version: 0.0.4
OS platform: win10

I try to import PR2 robot with URDF importer, it can be loaded sucessfully, but when I hit the play botton in the editor, it causes a crash.

The materials and meshes are from https://github.com/pr2/pr2_common
The urdf file has been tried with https://gist.github.com/drohr5D/5551267 and https://github.com/sehoonha/pydart2/blob/master/examples/data/urdf/PR2/pr2.urdf.
Both urdf file cause the crash issue.

vidurvij-Unity, 2020-12-30:
Hi @WenqiangX 
Thanks for using the URDF importer and bringing this to our attention. We are looking into the issue and will get back to you with a solution as soon as possible.
Vidur

vidurvij-Unity, 2021-01-21:
Hi @WenqiangX,
We reached out to the physics team and turns out right now we have a limit of 64 Articulation Bodies per GameObject. We are working on adding an error message if this happens in an imported robot, instead of the Editor crashing when play mode is entered. 
We are also working on supporting a higher number of Articulation Bodies per GameObject in the future.
Hope this allays any concerns you have in relation to this issue. Feel free to reach out if you have more questions.
Vidur

WenqiangX, 2021-01-22:
Thanks for your reply, hope the fix can be integrated into the Unity soon. @vidurvij-Unity

ashBabu, 2021-01-24:
Spent several days trying to fix the same issue with the 7-DOF  [Franka robot](https://github.com/frankaemika/franka_ros/tree/kinetic-devel/franka_description/robots) arm. Then I saw this. Not sure how to count this 64 articulation bodies per game object. A bit of explanation would be helpful.
Hope to see a solution soon.. Thanks

vidurvij-Unity, 2021-01-28:
@ashBabu Can you confirm which URDF file are you using in this repository ?

ashBabu, 2021-01-28:
I got the URDF after executing rosrun xacro --inorder panda_arm_hand.urdf.xacro

karthikm-0, 2021-02-20:
@ashBabu please see my post https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/125 if you are still having troubles.

vidurvij-Unity, 2021-03-16:
@WenqiangX and @ashBabu  An update has been added in Unity that warns if the imported robot has more than 64 joints. We will update the Articulation Body documentation in the near future.

safdark, 2021-03-20:
By when do you anticipate supporting *more than 64* bodies per game object? (Such as, 128?)

peifeng-unity, 2021-03-24:
Hi @safdark, thank you for asking. Our physics team is actively investigating to support more than 64 articulation body joints. It is a challenging problem due to the way that this cap is encoded in memory. We will keep you updated once there is more progress.

vidurvij-Unity, 2021-03-30:
@safdark The fix has been back ported to previous Unity versions. If you want to continue using v2021.1 I would recommend updating to 21.1.0b12 or any other newer Unity version. For more information you can reference https://issuetracker.unity3d.com/issues/editor-crashes-on-physx-ig-islandsim-activatenode-when-entering-play-mode-and-articulationbody-is-present-in-a-scene

safdark, 2021-04-27:
Thanks @vidurvij-Unity, and apologies for the delayed response. I tried the pre-release version 2021.2.0a14, but am seeing the same limit error:
"The maximum limit of ArticulationBody hierarchy size is reached: hierarchy of articulations can't have more than 64 ArticulationBody nodes!"

... But you mentioned that the "fix" has been made. Did I miss something?

closed 2021-05-10 by vidurvij-Unity