Unity-Technologies/URDF-Importer #32: URDF Robot switching axis spuriously

Th-Havy, 2021-01-12:
I encounter a bug with the URDF robot, which can occurs spuriously both in edit and play mode.

I am running Unity 2020.2.1f1.

The bug occurs when the field `Axis Type` of the `Urdf Robot` script is set to `Z Axis`, and it is shown in the following video:

https://user-images.githubusercontent.com/31824065/104349311-0c9fe380-5503-11eb-812c-d34552b1b326.mp4

Basically, this spurious axis switching can occur when the gameobject to which the UrdfRobot is attached is selected in the inspector.

I have traced the source of the bug in `UrdfRobot.cs`. Basically at line 26 there is:
```csharp
public ImportSettings.axisType choosenAxis ;
private ImportSettings.axisType currentOrientation = ImportSettings.axisType.yAxis;
```

But `currentOrientation` should instead be initialised with the value of `choosenAxis` [sic] e.g. in `OnEnable()` and/or in `Awake()`.

LaurieCheers-unity, 2021-01-12:
Thanks for the bug report, we'll look into this.

vidurvij-Unity, 2021-01-21:
Hi @Th-Havy,
This issue has been resolved in the latest commit.
https://github.com/Unity-Technologies/URDF-Importer/commit/68d91b3551f4c8073e03d6e5f48c5e909d121228
Vidur

mpinol, 2021-01-25:
Hi @Th-Havy,

I wanted to check in and see if Vidur's fix was able to resolve your issue?

Th-Havy, 2021-01-26:
Hi @mpinol ,

Thanks, it works well now, the bug has been corrected.
I'll close the issue.

closed 2021-01-26 by Th-Havy