Unity-Technologies/URDF-Importer #46: Adding Package From Git on Mac Fails

raeldor, 2021-02-07:
Pops up in the list briefly for less than a second then disappears.  Mac OS 11.2, Unity 2020.2.2f1.

rlunding, 2021-02-08:
I have the same problem on Mac and windows. Unity 2019.4.19f1

adverley, 2021-02-08:
Use Unity >= 2020.02.0

Clone this repo and make sure to init the submodules. Then, add package from disk. 

```
git clone git@github.com:Unity-Technologies/URDF-Importer.git && cd URDF-Importer
git checkout tags/v0.1.2
git submodule update --init --recursive
```

Tritanster, 2021-02-08:
I was getting the following error in the console: 

> [Package Manager Window] Cannot perform upm operation: Unable to add package [https://github.com/Unity-Technologies/URDF-Importer.git]:
  [https://github.com/Unity-Technologies/URDF-Importer.git] does not point to a valid package repository. No package manifest was found. Verify the repository URL and make sure the package is located in the root folder of the repository. [NotFound].
UnityEditor.EditorApplication:Internal_CallUpdateFunctions ()

The .json file is currently missing from the root of the repo. I think that's the issue. 

The suggested local checkout worked for me. Thank you!

vidurvij-Unity, 2021-02-08:
There were some files rearranged in the repo. We are updating its instructions. In the meantime you can use the git link for this repo with a release tag https://github.com/Unity-Technologies/URDF-Importer.git#v0.1.2 in the package manager.

I hope this solves your problem.
Vidur

raeldor, 2021-02-08:
This seemed to work.  I did get a couple of warnings, including...

Script 'Joint' has the same name as built-in Unity component.
AddComponent and GetComponent will not work with this script.

I assume this is by design?  Thanks.

vidurvij-Unity, 2021-02-08:
@raeldor Yes that correct.

raeldor, 2021-02-08:
Appreciate the quick response and the fix.  Thank you.

closed 2021-02-08 by raeldor