Unity-Technologies/URDF-Importer #47: Type or Namespace MeshProcess could not be found

raeldor, 2021-02-07:
After downloading the zip from github and installing the package by unzipping and selecting package.json I now get the following error...

/Users/xxx/Downloads/URDF-Importer-dev/com.unity.robotics.urdf-importer/Editor/NeedsRuntimeConversion/Extensions/UrdfGeometryCollision.cs(18,7): error CS0246: The type or namespace name 'MeshProcess' could not be found (are you missing a using directive or an assembly reference?)

vidurvij-Unity, 2021-02-08:
Hi @raeldor,
URDF importer has a submodule which is not installed when you download the zip file. I would recommend using this link https://github.com/Unity-Technologies/URDF-Importer.git#v0.1.2 in your package manager to install the package. This should resolve your issue. 
Vidur

raeldor, 2021-02-08:
This seemed to resolve the issue, thank you.

closed 2021-02-08 by raeldor