Unity-Technologies/URDF-Importer #59: Unstable behaviour with rigidbodies

mathujeyanth, 2021-02-24:
Hi, I'm encountering unstable behavior using the package together with rigidbodies. A minimal reproducible example can be seen with the articulation robotics project (https://github.com/Unity-Technologies/articulations-robot-demo).  
Picking up a cube with the package shows the following unstable behavior. The rotation and position keep changing (which eventually causes the cube to fall).  

[image]
Without the package, it is stable, with a constant rotation and minimal fluctuations in the position.  

[image]  

Decreasing the size of the rigidbodies increases the instability.

LaurieCheers-unity, 2021-03-02:
Hi, we're not seeing any significant difference in simulation stability with and without the URDF importer package. Are you sure that's the only difference between your two setups?

[image]

mathujeyanth, 2021-03-03:
Yes, there is no other difference than importing the URDF-importer tool. A video of the exact steps is shown in this [video](https://user-images.githubusercontent.com/40917493/109779746-8a7c9300-7c06-11eb-9876-b16655c9e0ba.mp4). I'm getting a warning with inconsistent line ending, but I'm really doubting that is the issue. I noticed the tool changed the rigidbody component, but I'm not sure if that has anything to do with it. So far I have been unable to solve the issue and any help would be appreciated.

LaurieCheers-unity, 2021-03-03:
Hi, thanks for the repro, I was able to follow your steps and get the same result.

It turns out this effect is caused by [RigidbodyEditor.cs](https://github.com/Unity-Technologies/URDF-Importer/blob/main/com.unity.robotics.urdf-importer/Editor/CustomEditors/RigidbodyEditor.cs): as long as you have a rigid body selected, its inertiaTensorRotation is constantly being converted to euler angles and back.

I made a fix and it will be released soon. To get it now, you can paste the following git URL into package manager:
https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#laurie/RigidbodyEditor

mathujeyanth, 2021-03-04:
Thank you for the fix! it now works as intended.

closed 2021-03-04 by mathujeyanth