Unity-Technologies/URDF-Importer #62: import Robot from Selected URDF file cannot be selected

MrVBian, 2021-03-10:
[image]
[image]

version: Unity2020.2.7f1c1

Thank you very much!

vidurvij-Unity, 2021-03-10:
Hey @MrVBian,
This is by design. Select the URDF file you want to import in your project view and do right click -> Import Robot from selected URDF File.

closed 2021-03-12 by MrVBian

Axin630, 2022-05-16:
> Hey @MrVBian, This is by design. Select the URDF file you want to import in your project view and do right click -> Import Robot from selected URDF File.

Why is ‘Import Robot from selected URDF File’ gray and can't be clicked

medhavikamran4, 2023-04-24:
What is the answer for this issue? I am facing same problem but it is not clear in the goven conversation , what is the procedure to resolve this issue.

hugoyvrn, 2023-06-30:
Be careful to click on the urdf file (since `.urdf` is hidden because the file extension may not appear in the Project window.).