Unity-Technologies/URDF-Importer #69: Hi,

Zimmo95, 2021-03-24:


Zimmo95, 2021-03-24:
Hi, 
After have followed the installation tutorial, I tried to import in Unity an already edited URDF file of a KUKA robot.  
When I upload the URDF file and the meshes in the Asset folder and push on the "import robot from URDF",  an error message appears telling me that Unity cannot find the asset file of each link in the specified path (picture) even though the file are located exactly in that path.
In fact, if i select by hand each asset file, unity finally opens correctly the robot.
Since it is a tedious operation, is there any way to avoid this problem? 
[image]

Thanks for the availability,
Enrico

peifeng-unity, 2021-03-25:
Hi Enrico, thank you for the feedback. We will take a look at it and come back to you.

vidurvij-Unity, 2021-04-07:
Can you share the file structure in your Project and your URDF file ?

Zimmo95, 2021-04-08:
[my_robot.zip](https://github.com/Unity-Technologies/URDF-Importer/files/6277437/my_robot.zip)

Best regards,
Enrico

at669, 2021-05-04:
Hi @Zimmo95, sorry for the delay! It looks like the URDF defined the mesh paths with `..` to go up a directory level, which was not being read correctly by the importer. Additionally, the `.urdf` file expects to be in the root of the URDF directory, as described in the [Tutorial Appendix](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md#file-hierarchy).

The reorganized version of your URDF directory is attached! I have replaced all references for the filenames from:
```xml
<mesh filename="../meshes/stl/Tip2 Scaled.stl" scale="1.15879 1.15879 1.15879"/>
```
to 
```xml
<mesh filename="package://meshes/stl/Tip2 Scaled.stl" scale="1.15879 1.15879 1.15879"/>
```
and I have also restructured the directory to look as follows:

```
my_robot
|-- meshes
    |-- stl
        |-- Link.002_0.asset
        |-- ...
|-- unnamed.urdf
```

[my_robot.zip](https://github.com/Unity-Technologies/URDF-Importer/files/6422646/my_robot.zip)

Please let me know if this resolves your issue! I will update the ticket for this issue to support parsing the `..` operator in a URDF filepath.

[ticket# AIRO-570]

Zimmo95, 2021-05-04:
Hi Amanda,
I downloaded and imported the folder in Unity and everything worked perfectly.
Thank you for your time and your availability!

Best regards,
Enrico

at669, 2021-05-04:
That's great to hear! The PR for the longer-term fix is linked above if you'd like to stay updated on that. In the meantime, I will close out this issue, but feel free to open a new one if another issue arises!

closed 2021-05-04 by at669