Unity-Technologies/URDF-Importer #70: Invalid AABB errors still exists in Unity 2020.3 with URDF-importer 0.2.0

WenqiangX, 2021-04-01:
I found this [issue](https://github.com/Unity-Technologies/URDF-Importer/issues/11), and I tried the solution with no use. Since it is closed, I think it might be ignored, so I open a new issue.

For the Franka Panda arm, I found the 'disable collision' cannot fix the Invalid AABB errors.

Some relevant information:
1. the Franka panda arm urdf can be loaded successfully in Pybullet.
2. Unity version is 2020.2 and 2020.3, URDF importer 0.2.0, both have invalid AABB errors.
3. When I use URDF importer 0.0.5 in Unity 2020.3, the MessProcess cannot be found.
Can anyone help? Thanks in advance!

vidurvij-Unity, 2021-04-02:
The fix is still in the dev branch, In the mean time, you can use the following release
https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#v0.2.1

hyounesy, 2021-04-10:
Hello @WenqiangX , just following up to see if using the v0.2.1 help solving the issue above.

vidurvij-Unity, 2021-05-10:
Hi @WenqiangX. The issue should be solved in the latest version of URDF Importer. Please feel free to reopen the issue if the problem persists.

closed 2021-05-10 by vidurvij-Unity