Unity-Technologies/URDF-Importer #71: Model and collision rotation mismatch

Th-Havy, 2021-04-06:
Hi,

I am trying to load an URDF, composed of `.obj` models, but using only primitive shape for collisions (Cylinders and spheres). The model is as shown in the figure below:
[image]
_URDF Model visualized in Rviz_

The URDF importer loads the model correctly, but the rotation of the visuals and collision does not match:
[image]
_Axis type_ `Y axis`

[image]
_Axis type_ `Z axis`

As you can see, the colliders are correct in `Y Axis`, while the visuals are correct in `Z Axis`. After quite some debugging and experimentation, I have realized the problem was the following: The primitive collisions (and similarly for the primitive visuals) should not have their orientation corrected, because they are create in Unity, using Unity's coordinate system rather than ROS'. Hence their rotation is already correct.

peifeng-unity, 2021-04-06:
Thank you @Th-Havy for letting us know about this issue. We will work on fixing it. In the meantime, the simplest workaround is to manually change the rotation of Unity's game objects (either collider or visual) in the editor to align them. Thanks again for your feedback.

Th-Havy, 2021-04-07:
Thanks a lot for your reply, simply flipping back the model parts is sufficient as a workaround.

As a side note, it would be nice to add the axis flipping to the undo/redo stack.

In case it can help anyone I have created a very basic script to flip either the visuals or the collisions, just as it is done in `CorrectAxis()` of `URDFRobotExtensions.cs`. Attach the following script to a gameobject with an `URDFRobot` component and you can flip the parts as required:

```csharp
using System.Collections;
using System.Collections.Generic;
using UnityEngine;

/// <summary>
///  Attach this script to a URDFRobot gameobject for which you wish to flip the collisions or visuals.
/// For simplicity no custom editor is created, but functions are simply called when tickboxes are ticked.
/// </summary>
[ExecuteInEditMode]
public class FlipRobot : MonoBehaviour
{
    public enum FlippedObjects
    {
        Visuals,
        Collisions
    }

    [Tooltip("Select which objects to flip in the URDFRobot.")]
    public FlippedObjects objectsToFlip;
    [Tooltip("Correct the visuals to the selected axis type.")]
    public RosSharp.ImportSettings.axisType axisCorrection;
    [Tooltip("Tick this to flip the objects.")]
    public bool flipObjects;

    void Update()
    {
        if (flipObjects)
        {
            Flip();
        }
        
        flipObjects = false;        
    }

    void Flip()
    {
        Quaternion correctYtoZ = Quaternion.Euler(-90, 0, 90);
        Quaternion correctZtoY = Quaternion.Inverse((correctYtoZ));
        Quaternion correction = new Quaternion();

        if (axisCorrection == RosSharp.ImportSettings.axisType.zAxis)
            correction = correctYtoZ;
        else
            correction = correctZtoY;

        // Register rotations for undo system
        UnityEditor.Undo.IncrementCurrentGroup();
        UnityEditor.Undo.SetCurrentGroupName("Flip Objects");
        int undoGroupIndex = UnityEditor.Undo.GetCurrentGroup();

        if (objectsToFlip == FlippedObjects.Visuals)
        {
            RosSharp.Urdf.UrdfVisual[] visualMeshList = gameObject.GetComponentsInChildren<RosSharp.Urdf.UrdfVisual>();

            foreach (RosSharp.Urdf.UrdfVisual visual in visualMeshList)
            {
                UnityEditor.Undo.RecordObject(visual.transform, "Flip URDF visuals");
                visual.transform.localRotation = visual.transform.localRotation * correction;
            }
        }
        else
        {
            RosSharp.Urdf.UrdfCollision[] collisionMeshList = gameObject.GetComponentsInChildren<RosSharp.Urdf.UrdfCollision>();

            foreach (RosSharp.Urdf.UrdfCollision collision in collisionMeshList)
            {
                UnityEditor.Undo.RecordObject(collision.transform, "Flip URDF collisions");
                collision.transform.localRotation = collision.transform.localRotation * correction;
            }
        }

        // Make all rotations a single undo operation
        UnityEditor.Undo.CollapseUndoOperations(undoGroupIndex);
    }
}
```

peifeng-unity, 2021-04-14:
Hi @Th-Havy, I was trying to reproduce the issue with [the Husky robot](https://github.com/husky/husky/blob/melodic-devel/husky_description/urdf/husky.urdf.xacro) as an example, which uses box/cylinder as colliders and .dae mesh as visual objects, but the visual and collision are matching with each other. Would you mind sharing your URDF assets and it may help us to reproduce the issue? Thank you in advance!

Th-Havy, 2021-04-15:
Hey @peifeng-unity, I have attached the urdf and meshes of my robot which shows this behavior. However I'm using the `.obj` format and not `.dae`.
[vacuum_robot.zip](https://github.com/Unity-Technologies/URDF-Importer/files/6318689/vacuum_robot.zip)

vidurvij-Unity, 2021-05-10:
This issue has been solved and the fix is available in the latest version of the URDF Importer.
Vidur

closed 2021-05-10 by vidurvij-Unity

eliphatfs, 2025-05-22:
[image]
seems i am still encountering this issue with the latest version (urdf: https://github.com/haosulab/ManiSkill/tree/main/mani_skill/assets/robots/panda panda_v3.urdf)