Unity-Technologies/URDF-Importer #74: Rotation Drift of Rigidbody while Grasping

mathujeyanth, 2021-04-12:
Hi, I'm experiencing unstable behaviour when grasping rigidbodies with an imported gripper using v0.2.0 of the importer. This issue is maybe related to [#59](https://github.com/Unity-Technologies/URDF-Importer/issues/59).
I'm importing the Hand-E gripper from https://github.com/AcutronicRobotics/robotiq_modular_gripper.git and trying to grasp a simple cube, but the cube keeps drifting in its rotation as in [#59](https://github.com/Unity-Technologies/URDF-Importer/issues/59).
I have tried adjusting the physics settings, but without much luck.
Below is a simple reproduction:
[image]

hyounesy, 2021-04-12:
Thanks for bringing this to our attention. Our initial guess is that it might be related to intra mesh collision in the robot. We will investigate this and get back to you.
[ticket# AIRO-597]

at669, 2021-05-06:
Hi @mathujeyanth! After some investigation, I was able to reproduce this issue. The key issue is as @hyounesy mentioned--the right and left fingers of the gripper would collide internally when gripping small objects like the 0.01 scaled cube. I'm attaching a video showing the intra-mesh collisions, and the stopped drifting!

https://user-images.githubusercontent.com/31416491/117335045-f1f9cd80-ae57-11eb-8ec2-007ddfd791f6.mov

These were the changes I made in order to stop the rotation drifting:

1. Disabling collisions between the right and left fingers in the URDF file resolves the root of the issue. In the `robotiq_hande.urdf` file, add the following:
    ```xml
    <disable_collision link1="hande_right_finger" link2="hande_left_finger"></disable_collision>
    ```
    You can learn more about disabling collisions [here](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/urdf_importer/urdf_appendix.md#disable-collision-support).
2. I see that you've already selected TGS as the solver type, which is great! There are two options here:
    1. if you'd like to use the `Two Directional Friction Type`, you'll need to increase the `Default Solver Iterations` to, for example, 12, as directional friction requires more iterations to compute.
    2. If you keep the `Patch Friction Type`, you can leave the `Default Solver Iterations` to the default of 6.
3. For either option, I left the `Default Solver Velocity Iterations` as the default for the video sample taken, but you're free to tweak the values as you see fit!

Here are my physics settings for reference:
[image]

Finally, here is the modified URDF: [robotiq_hande.urdf.zip](https://github.com/Unity-Technologies/URDF-Importer/files/6436311/robotiq_hande.urdf.zip)

Please let us know if these steps resolve your issue!

mathujeyanth, 2021-05-06:
Hi @at669 thank you for the reply! I did find a (somewhat) solution to the problem, but forgot to update this issue and resolve it. I first arrived at the same conclusion that the drift was caused by the collision of the right and left fingers collider, so I manually disabled the responsible colliders and had to tune some of the physics value. Though I prefer your solution as it being more elegant. The linked URDF Tutorials Appendix is also much appreciated!
Thank you for the solution, I'll resolve this issue now.