Unity-Technologies/URDF-Importer #75: Missing `Constraints` field in RigidBody inspector

Th-Havy, 2021-04-12:
Hi,

The custom inspector implemented in this package for rigidbodies is not showing the rigidbody constraints.
This is quite annoying, as the only way to set the constraints is then to use scripts.
[image]
_Default rigidbody inspector_

[image]
_Custom rigidbody inspector missing the constraint options_

A quick workaround is to modify the custom editor as follows:
```csharp
// Insert at line 16 of RigidvodyEditor.cs
_rigidbody.constraints = (RigidbodyConstraints)EditorGUILayout.EnumFlagsField("Constraints", _rigidbody.constraints);
```

However, please note that this will not produce the original formatting, but the one below. Maybe you guys can get the original formatting from the original rigidbody inspector source code.
[image]

hyounesy, 2021-04-12:
Thank you @Th-Havy for bringing this to our attention and suggesting this feature. We have added a ticket to investigate a proper approach to add this to the custom inspector. 
[ticket#: AIRO-598]

at669, 2021-06-21:
Hi @Th-Havy, sorry for the late reply--I believe this issue was resolved with the URDF-Importer [v0.3.0 update](https://github.com/Unity-Technologies/URDF-Importer/pull/86), which removed the custom editor for Rigidbodies (however, we are now on v0.4.0, which I suggest updating to!). Sorry for the inconvenience. I will close this ticket as the issue has been resolved, but feel free to reopen it if the issue persists. Thanks!

closed 2021-06-21 by at669