Unity-Technologies/URDF-Importer #76: namespace name 'ArticulationBody' could not be found

masum74, 2021-04-14:
I'm getting the following error while adding the URDF package using the git URL:

Library\PackageCache\com.unity.robotics.urdf-importer@5ba94b3aac\Runtime\Controller\JointControl.cs(15,12): error CS0246: The type or namespace name 'ArticulationBody' could not be found (are you missing a using directive or an assembly reference?)

Can you please help me to resolve this issue? Thanks!

peifeng-unity, 2021-04-14:
Hello @masum74, what is the Unity version that you are using? The Unity `ArticulationBody` component is a new feature that was introduced in Unity 2020.1+.

masum74, 2021-04-14:
Hey! I upgraded Unity from 2019 to 2021 and the problem resolved. Thanks!

closed 2021-04-14 by masum74

Gregory-Baker, 2021-08-05:
Hi, does anyone know if there is any way to add the `ArticulationBody` class to 2019.4.x? I'm constrained to use this version by another package I'm using unfortunately...