Unity-Technologies/URDF-Importer #77: Velocity control affected by gravity

mathujeyanth, 2021-04-14:
Hi, I'm attempting to implement a velocity control and following issue #24 the stiffness should be set to 0, but by doing so it is affected by gravity. The gif below illustrate what I'm talking about. 

[image]
I would expect the gripper to be standing still and only move based on the specified velocity targets, where none is given here.

The reproduction setup is using https://github.com/Unity-Technologies/articulations-robot-demo, where I have only modified the FixedUpdate() in GripperDemoController.cs to be the following:
```cs
  private void FixedUpdate()
  {
      ArticulationBody articulation = GetComponent<ArticulationBody>();
      float targetVelocity = -(float)moveState * Time.fixedDeltaTime * speed;
      var drive = articulation.xDrive;
      drive.targetVelocity = targetVelocity;
      articulation.xDrive = drive;
  }
```

How would I go about this, without changing the physics setting for the cube e.g turning off the cube's gravity?

peifeng-unity, 2021-04-14:
Hello @mathujeyanth, the behavior is expected when you drive the articulation body with the target velocity and setting the stiffness to zero. The articulation body is purely velocity-driven in this case and the force is calculated by the difference between the target velocity and the actual velocity ([see the formula here](https://docs.unity3d.com/2020.1/Documentation/ScriptReference/ArticulationDrive.html)). That is saying the gripper is expected to continue moving under some forces even the target velocity is set to zero by the script.

There are a few ways to make it work as you expected. The simplest one is to set the articulation body a fixed joint when it should be still. I attached the modified FixedUpdate() below for your reference. Please let us know if it works for you.

```
private void FixedUpdate()
{
    ArticulationBody articulation = GetComponent<ArticulationBody>();
    var drive = articulation.xDrive;

    if (moveState == BigHandState.Fixed)
    {
        articulation.jointType = ArticulationJointType.FixedJoint;
        drive.targetVelocity = 0;
        articulation.xDrive = drive;
        return;
    }

    articulation.jointType = ArticulationJointType.PrismaticJoint;
    float targetVelocity = -(float)moveState * Time.fixedDeltaTime * speed;
    drive.targetVelocity = targetVelocity;
    articulation.xDrive = drive;

    gameObject.SetActive(false);
    gameObject.SetActive(true);
}
```

https://user-images.githubusercontent.com/56408141/114777152-2fbb7880-9d28-11eb-992b-1a9adbf581c6.mov

mathujeyanth, 2021-04-15:
Thank you @peifeng-unity for the quick reply! That solution seems to work. I'll close this issue, but could you explain to me why the `SetActive` is necessary?

peifeng-unity, 2021-04-15:
It is great to hear that this solution works for you! For the question, you may notice that the articulation body uses [reduced coordinates with the Featherstone solver](https://docs.unity3d.com/2020.1/Documentation/ScriptReference/ArticulationBody.html) to calculate the dynamics of robot joints. The coordinates are changed when one of the joint types gets modified and the physics engine needs to reload the list of joints and create the correct coordinates.

For example, the degree of freedom (DOF) for the fixed joint is zero and the DOF for the prismatic joint is one. When we change the joint type from fixed to prismatic during the runtime, the reduced coordinates are modified and need to be rebuilt since there is one more DOF. The `SetActive` commands are going to reactivate all the ArticulationBody components on the gripper and its children objects, which resets the coordinates and makes the joints work correctly.

Please feel free to let us know if it makes sense to you.

mathujeyanth, 2021-04-16:
Thank you, that makes sense! I'll close this issue.

mathujeyanth, 2021-04-19:
Hi, I'm reopening this issue, since the solution has issues grasping while moving. Using the same configuration, moving up and down makes the cube eventually slip out. This is even more visible when decreasing the size of the cube, where none of these issue are seen when using position control. 
Here using velocity control:

https://user-images.githubusercontent.com/40917493/115218351-c3b18980-a106-11eb-8f09-4bfcda87901f.mp4

Using position control (with same setup except stiffness set to 100.000)

https://user-images.githubusercontent.com/40917493/115218478-e17eee80-a106-11eb-8fce-1d8b44e350d9.mp4

I'm suspecting it to be expected behaviour related to the explanation from @peifeng-unity:
> Hello @mathujeyanth, the behavior is expected when you drive the articulation body with the target velocity and setting the stiffness to zero. The articulation body is purely velocity-driven in this [...]
But how do I mitigate this?

peifeng-unity, 2021-04-26:
Hi @mathujeyanth, please try the following code. Please let us know if this migration works for you.

```
private BigHandState lastMoveState = BigHandState.Fixed;

private void FixedUpdate()
{
    ArticulationBody articulation = GetComponent<ArticulationBody>();
    var drive = articulation.xDrive;

    if (moveState != lastMoveState)
    {
        lastMoveState = moveState;
        gameObject.transform.parent.gameObject.SetActive(false);
        gameObject.transform.parent.gameObject.SetActive(true);
    }

    if (moveState == BigHandState.Fixed)
    {
        articulation.jointType = ArticulationJointType.FixedJoint;
        drive.targetVelocity = 0;
        articulation.xDrive = drive;
    }
    else
    {
        articulation.jointType = ArticulationJointType.PrismaticJoint;
        float targetVelocity = -(float)moveState * Time.fixedDeltaTime * speed;
        drive.targetVelocity = targetVelocity;
        articulation.xDrive = drive;
    }
}
```

mathujeyanth, 2021-04-27:
Thank you very much @peifeng-unity, that works!

closed 2021-04-27 by mathujeyanth