Unity-Technologies/URDF-Importer #79: Issue Exporting Box Collider

micahpearlman, 2021-04-15:
Box Collider primitive does not export size.  It actually exports the transform scale.   

This is how the collisions are setup:

[image]

And the Box Collider definition:

[image]

And then the resulting URDF:

```
    <collision>
      <geometry>
        <box size="1 1 1" />
      </geometry>
    </collision>
    <collision>
      <geometry>
        <box size="1 1 1" />
      </geometry>
    </collision>
    <collision>
      <geometry>
        <box size="1 1 1" />
      </geometry>
```

LaurieCheers-unity, 2021-04-20:
Hi, thanks for bringing this to our attention. I've made a ticket to track the issue internally. [AIRO-623]

vidurvij-Unity, 2021-05-12:
Hi @micahpearlman,
Can you share the URDF file you are trying to import. I checked the code with an example and it seems the export function is working correctly. The URDF importer does not use the size variables in the collider to change of the primitives. Instead it changes the scale of the gameObject to obtain the desired size, which is what it exports as the size of the primitive which in this case is a box.

micahpearlman, 2021-05-12:
@vidurvij-Unity  I have since moved on to other solutions to export URDF and have deleted the test project.  I had submitted a pull request that fixed this issue but found the process to submit a PR to Unity arduous and have since deleted the PR.  The best example available is still what is outlined above.  Given a Unity Box Collider with Size set, the collision URDF export box sizes are all size "1 1 1".  If that is not the case anymore then feel free to close this issue.

vidurvij-Unity, 2021-05-13:
@micahpearlman  We would have loved if you would have continued with the usage of this repository. The issue raised here was because of lack of documentation. When we import a URDF file, we use scale of the parent to change the size of the primitive mesh. So if you want to export the size of the 'box' you created, you can change the scale of the parent to export the size to the URDF. Its working and ready to use. Feel free to give it a try again and we hope you return to use this repo.

vidurvij-Unity, 2021-07-06:
The updated documentation on primitive sizing can be found here.
https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/dev/tutorials/urdf_importer/urdf_appendix.md##Sizing-of-Primitives

closed 2021-07-15 by vidurvij-Unity