Unity-Technologies/URDF-Importer #81: Can't add component 'ArticulationBody' to base because such a component is already added to the game object!

Th-Havy, 2021-04-19:
Hi,

When importing URDFs, I often times get the following log several times on the console. It is no big deal and does not prevent the importation, but it is annoying to have to clear the console output.

```
Can't add component 'ArticulationBody' to base because such a component is already added to the game object!
UnityEngine.GameObject:AddComponent<UnityEngine.ArticulationBody> ()
RosSharp.Urdf.UrdfJointFixed:Create (UnityEngine.GameObject) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Runtime/UrdfComponents/UrdfJoints/UrdfJointFixed.cs:27)
RosSharp.Urdf.UrdfJoint:AddCorrectJointType (UnityEngine.GameObject,RosSharp.Urdf.UrdfJoint/JointTypes) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Runtime/UrdfComponents/UrdfJoints/UrdfJoint.cs:80)
RosSharp.Urdf.UrdfJoint:Create (UnityEngine.GameObject,RosSharp.Urdf.UrdfJoint/JointTypes,RosSharp.Urdf.Joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Runtime/UrdfComponents/UrdfJoints/UrdfJoint.cs:64)
RosSharp.Urdf.Editor.UrdfLinkExtensions:ImportLinkData (RosSharp.Urdf.UrdfLink,RosSharp.Urdf.Link,RosSharp.Urdf.Joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfLinkExtensions.cs:57)
RosSharp.Urdf.Editor.UrdfLinkExtensions:Create (UnityEngine.Transform,RosSharp.Urdf.Link,RosSharp.Urdf.Joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfLinkExtensions.cs:31)
RosSharp.Urdf.Editor.UrdfLinkExtensions:ImportLinkData (RosSharp.Urdf.UrdfLink,RosSharp.Urdf.Link,RosSharp.Urdf.Joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfLinkExtensions.cs:65)
RosSharp.Urdf.Editor.UrdfLinkExtensions:Create (UnityEngine.Transform,RosSharp.Urdf.Link,RosSharp.Urdf.Joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfLinkExtensions.cs:31)
RosSharp.Urdf.Editor.UrdfRobotExtensions:Create (string,RosSharp.ImportSettings) (at Library/PackageCache/com.unity.robotics.urdf-importer@5ba94b3aac/Editor/NeedsRuntimeConversion/Extensions/UrdfRobotExtensions.cs:74)
URDFImporterEditor:OnInspectorGUI () (at Assets/vacuum_robot/Scripts/Custom Editors/URDFImporterEditor.cs:62)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)
```

I believe the problem comes from the fact that `AddComponent<ArticulationBody>()` is called, when a previously added component has a `[RequireComponent(typeof(ArticulationBody))]` attribute.

LaurieCheers-unity, 2021-04-21:
Just released version 0.3.0, which should fix this.

Th-Havy, 2021-04-22:
Indeed, the warning is gone in 0.3.0, thanks a lot.
I'll close the issue then.

closed 2021-04-22 by Th-Havy