Unity-Technologies/URDF-Importer #113: Cannot click Import Robot from Selected URDF File

madd-aj, 2021-05-30:
[image]

I tried right-clicking it as I saw someone else said, but that doesn't work either. Not sure what's going on here.
Using Unity 2020.2.0b93373

Thank you!

peifeng-unity, 2021-06-01:
Hello @madd-aj, sorry for the late reply. You may need to right-click on the [`niryo_one.urdf`](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/PickAndPlaceProject/Assets/URDF/niryo_one/niryo_one.urdf) file inside the `niryo_one` folder. Please let us know if this could help to solve your issue.

at669, 2021-06-21:
Hi all, I will be closing this ticket as the solution has been provided and also due to inactivity. Please feel free to reopen the ticket if the problem persists. Thanks!

closed 2021-06-21 by at669