Unity-Technologies/URDF-Importer #121: createStlPrefab overwriting existing prefabs

robin-moss, 2021-06-29:
**Describe the bug**

When unity is started or reimports an STL asset the URDF importer detects we have some STL files in our project (unrelated to any URDFs) and creates a prefab despite one already existing. This overwrites the existing prefab.

**To Reproduce**
1. Install URDF importer
2. Create a folder called `assets`
2. Put an STL file in that folder
3. Use the STL file to create a prefab with material
4. Save all the changes
5. Asset -> Reimport
6. The scene goes very very pink

**Console logs / stack traces**

```
Detected an stl file at Assets/.../....stl - creating a mesh prefab.
UnityEngine.Debug:Log (object)
RosSharp.StlAssetPostProcessor:OnPostprocessAllAssets (string[],string[],string[],string[]) (at Library/PackageCache/com.unity.robotics.urdf-importer@d90212e1cd/Runtime/MeshProcessing/StlAssetPostProcessor.cs:39)
UnityEditor.AssetPostprocessingInternal:PostprocessAllAssets (string[],string[],string[],string[],string[]) (at /home/bokken/buildslave/unity/build/Editor/Mono/AssetPostprocessor.cs:167)
```

**Expected behavior**
If a prefab already exists do not overwrite it

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.12fl
- Unity machine OS + version: Ubuntu 20
- ROS machine OS + version: Ubuntu 20 + Noetic (not ros used for this)
- ROS–Unity communication: N/A
- Branch or version: v0.4.1

**Additional context**

vidurvij-Unity, 2021-06-29:
Thank you for bringing this to our attention. I have created a ticket internally to track it. We will update you when this has been patched.

hyounesy, 2021-07-07:
ticket#: AIRO-908

hyounesy, 2021-08-03:
The existing behaviour of URDF Importer where it was overwriting existing assets in a Unity project has now changed.

As of 0.5:
-   No assets will be created for .stl meshes in the postprocess step.
-   Instead, the assets will be created during URDF Import if they don't exist already.
-   A checkbox is added to the URDF Import UI to allow users to recreate the assets for STL files even if they already exist.

closed 2021-08-03 by hyounesy