Unity-Technologies/URDF-Importer #122: URDF Importer causing build errors in Unity

srujan-aris, 2021-06-29:
I'm using URDF importer tool to import my ROS env into unity. The import is successful. I could also simulate the environment and everything looks to be working fine. However when I try building the unity project for UWP I get build errors. Looks like these errors are related to URDF Importer. When I remove the URDF importer from the package manager, the build succeeds with no errors, but I think it leaves the unity project unstable

Here's the console output
```
Plugin 'assimp.dll' is used from several locations:
 Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll would be copied to <PluginPath>/assimp.dll
 Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll would be copied to <PluginPath>/assimp.dll
Please fix plugin settings and try again.

UnityEditor.Modules.DefaultPluginImporterExtension:CheckFileCollisions (string)
UnityEditorInternal.PluginsHelper:CheckFileCollisions (UnityEditor.BuildTarget)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)
```

Here's my build options. I will be installing this application to Hololens2

[image]

**Environment**
- Unity Version: Unity 2020.3.12f1
- Unity machine OS + version: Windows 10
- ROS machine OS + version: Ubuntu 20.04, ROS Noetic
- Branch or version: [e.g. v0.4.0]

vidurvij-Unity, 2021-07-01:
Hi,
Thanks for bringing this to our attention. We will look into this error and get back to you with a potential solution.

hyounesy, 2021-07-10:
Ticket# AIRO-920

sdiao, 2021-08-02:
Closing this issue since the fix has been merged in https://github.com/Unity-Technologies/URDF-Importer/pull/124

closed 2021-08-02 by sdiao