Unity-Technologies/URDF-Importer #134: URDF Robot Base Joint Falls or Falls Apart

Pterodactylus, 2021-07-18:
**Describe the bug**
When I import a generic URDF file the link above the base link falls through the simulation or falls off to the side. I've replicated this with more than one urdf file. Any help would be appreciated.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new blank scene
2. Go to attached pi_robot.urdf file and import the robot as per the normal instructions
3. Click the play button to start the scene

**Expected behavior**
Robot should hold together link in the scene window.
[image]

**Actual undesired behavior**
Robot falls through the base link like this.
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.14f1 & 2021.2.0b4
- Unity machine OS + version: Windows 10 & Ubuntu 18.04
- ROS machine OS + version: N/A
- ROS–Unity communication: N/A
- Branch or version: 0.5.0

**URDF File**
`<robot name="pi_robot">

	

 	<link name="base_link">
		<visual>
		    <origin xyz="0 0 0.0425" rpy="0 0 0"/>
			<geometry>
				<box size="0.32 0.26 0.085"/>
			</geometry>
			<material name="Cyan1">
	       		<color rgba="0 0.9 0.9 1.0"/>
	     	</material>
		</visual>	
	</link>
	
 	<link name="base_laser">
		<visual>
		    <origin xyz="0 0 0" rpy="0 0 0"/>
			<geometry>
				<cylinder radius="0.025" length="0.07"/>
			</geometry>
			<material name="Black1">
	       		<color rgba="0.2 0.2 0.2 1.0"/>
	     	</material>
		</visual>	
	</link>
	
	<link name="cpu_link">
		<visual>
		    <origin xyz="0 0 0.035" rpy="0 0 0"/>
			<geometry>
				<box size="0.19 0.19 0.07"/>
			</geometry>
			<material name="Cyan2">
	       		<color rgba="0 0.7 0.7 1.0"/>
	     	</material>
		</visual>	
	</link>

  	<link name="upper_base_link">
		<visual>
			<origin xyz="0 0 0.05" rpy="0 0 0"/>
			<geometry>
				<cylinder radius="0.085" length="0.10"/>
[... 424 lines cut ...]
       	<limit lower="-1.57" upper="1.57" effort="10" velocity="3"/>
        <axis xyz="1 0 0"/>
	</joint>
	
 	<joint name="right_wrist_joint" type="revolute">
    	<parent link="right_lower_arm_link"/>
    	<child link="right_wrist_link"/>
    	<origin xyz="0 0 -0.05" rpy="0 0 0"/>
       	<limit lower="-1.57" upper="1.57" effort="10" velocity="3"/>
        <axis xyz="1 0 0"/>
	</joint>
	
 	<joint name="left_hand_joint" type="fixed">
    	<parent link="left_wrist_link"/>
    	<child link="left_hand_link"/>
    	<origin xyz="0 0 -0.055" rpy="0 0 0"/>
	</joint>
	
 	<joint name="right_hand_joint" type="fixed">
    	<parent link="right_wrist_link"/>
    	<child link="right_hand_link"/>
    	<origin xyz="0 0 -0.055" rpy="0 0 0"/>
	</joint>
 
</robot>`

peifeng-unity, 2021-07-19:
Hello @Pterodactylus, thanks for your reporting. This seems a duplicate of [another open issue in Unity-Robotics-Hub](https://github.com/Unity-Technologies/Unity-Robotics-Hub/issues/262). We will take a look and reply soon.

mrpropellers, 2021-08-04:
Closing this one to consolidate discussion

closed 2021-08-04 by mrpropellers