Unity-Technologies/URDF-Importer #135: Missing collision meshes references in generated mesh colliders

thibnoel, 2021-07-21:
**Describe the bug**
Hello,
I noticed that using the URDF importer with a URDF model containing multiple references to the same collision mesh leads to missing mesh colliders in the generated Unity model for all links sharing this reference except one (specifically, the mesh colliders under ArticulationBody/Collisions/link are generated correctly but without a reference to the correct mesh)

**To Reproduce**
Steps to reproduce the behavior:
1. Download a URDF with multiple ref. to a same collision mesh (the PR2 for example : https://github.com/PR2/pr2_common/tree/melodic-devel/pr2_description) and the associated meshes
2. Use the URDF Importer as described in the tutorials
3. Check the wheels ArticulationBody in the editor : one clearly displays the mesh collider, not the other
4. On play, one of the wheels goes through the ground

**Console logs / stack traces**
Not applicable

**Expected behavior**
All ArticulationBody generated by the URDF Importer have correctly referenced meshes for their children mesh colliders.

**Screenshots**
Example with a 2 armed robot : 
Left arm with correct reference to the collision meshes : 
[image]
Right arm with correct hierarchy generation but unreferenced collision meshes : 
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2021.2.0b1
- Unity machine OS + version: Ubuntu 20.04
- ROS machine OS + version: Ubuntu 18.04, ROS melodic
- ROS–Unity communication: ROS-TCP endpoint
- Branch or version: v0.5.0

**Additional context**
None

LaurieCheers-unity, 2021-07-23:
Thanks. I've filed an internal ticket to investigate this issue.

at669, 2021-09-27:
The PR for this issue has been merged into `dev` and will be in the next release of the URDF-Importer. Please feel free to try it out on `dev`. I'll be closing this issue, but feel free to reopen it if the problem persists, or open a new ticket if you encounter other issues.

closed 2021-09-27 by at669