Unity-Technologies/URDF-Importer #140: Unable to import URDF files with OBJ meshes at runtime.

707346129, 2021-09-07:
I can import URDF files with OBJ meshes through Unity GUI, but when I tried to import them at runtime with the UrdfRobotExtensions.CreateRuntime() method, I would meet errors like "Unable to create mesh collider for the mesh: package://textured_objs/original-7.obj". I checked the code at URDF-Importer-0.5.0/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfGeometryCollision.cs:93 and find that it probably only supports STL meshes. Is it possible to import URDF files with OBJ meshes at runtime?

hyounesy, 2021-09-07:
[Internal Ticket#: AIRO-1212]

peifeng-unity, 2021-09-07:
Hi @707346129, thanks for letting us know. Just wish to double-check whether you are trying to import URDF files in Unity Editor play mode or in a standalone executable? If it is in Unity Editor play mode, you could use the `UrdfRobotExtensions.Create()` method to import, which supports OBJ meshes. For the standalone executable, currently, it doesn't support importing OBJ meshes, but we could add this feature. Please let us know.