Unity-Technologies/URDF-Importer #143: How to manually change the transform of base link at runtime?

707346129, 2021-09-10:
I want to change the position of the base link at runtime, but when I tried to set the value of transform.position of base link, it would go back to the original point at next frame. Is there any method to manually change the transform of base link at runtime?

peifeng-unity, 2021-09-10:
Hi, this is the expected behaviour and the `ArticulationBody` component on the base link makes it not possible to manually change the transform by modifying the `transform.position`. To move the base link, you could try to use `ArticulationBody.xDrive.target` to drive the object with a force or teleport the base link via `ArticulationBody.teleportRoot`.

A-Ivan, 2021-09-10:
@peifeng-unity, would another workaround to this be to have the robot as a child of another gameobject (i.e., an empty gameobject or a moving platform), and change the `transform.position` of that parent gameobject?

peifeng-unity, 2021-09-10:
Ah, thanks @A-Ivan for reminding me! Yes, it would also work if the robot is a child of another game object. It would work with no problem if the moving platform also has articulation body component attach. Otherwise, the `immovable` flag needs to be unchecked on the base link articulation body so that the robot can move with the gameobject. However, the robot may fall in gravity if the configuration is not symmetric (e.g. a robot arm. It would not be a problem for a mobile robot). To connect the base link and the moving platform, you may also need to add [`configurable joint`](https://docs.unity3d.com/2020.3/Documentation/Manual/class-ConfigurableJoint.html), which can anchor the base link on the platform.

at669, 2021-11-30:
Hi all, since the answer has been provided above, I will close out this issue. Please feel free to reopen if there is any follow-up.

closed 2021-11-30 by at669

wyd0817, 2023-06-19:
Hello everyone, I've encountered a similar problem. I'm trying to run my environment within the ML-Agent setting. However, I've noticed that my robot is unable to promptly return to the position I've assigned. My robot is a c30r tracked robot, named 'robot'. There's no ArticulationBody or RigidBody present on 'robot'. However, there is an ArticulationBody (excluding xDrive) on /c30r/base_link and also ArticulationBody (including xDrive) on the child objects of /c30r/base_link.

Here, robot/base_link serves as the root node for the ArticulationBody, which doesn't possess an xDrive due to the absence of a parent ArticulationBody for it to rotate or move in relation to. As for the child objects of robot/base_link, they do have the ArticulationBody component, including the xDrive. This is because they may need to rotate or move relative to their parent object, which is the robot/base_link.
[image]

In the Initialize() function, I keep track of the robot's state, and attempt to restore the robot's orientation and position in the OnEpisodeBegin(). Despite following the advice given above, I'm failing. Below is the core of my code, am I missing something?
```sharp
public class PathGeneratorAgent : Agent
{
    private List<(ArticulationBody, Vector3, Quaternion)> initialStates;
    private Vector3 initialRobotPosition;
    private Quaternion initialRobotRotation;

	  public override void Initialize() 
    {   
        GetInitialStates();  
    }

   
    public override void OnEpisodeBegin()
    {
        ResetToInitialState();
    }

    public void GetInitialStates()
    {
        // Store the initial state of the robot
        initialRobotPosition = robot.transform.position;
        initialRobotRotation = robot.transform.rotation;

        // Store the initial state of all ArticulationBodies in the robot
        initialStates = new List<(ArticulationBody, Vector3, Quaternion)>();
        ArticulationBody[] articulationBodies = robot.GetComponentsInChildren<ArticulationBody>();
        foreach (ArticulationBody body in articulationBodies)
        {
            initialStates.Add((body, body.transform.localPosition, body.transform.localRotation));
        }
    }

    public void ResetToInitialState()
    {
        // Reset the robot to its initial state
        robot.transform.position = initialRobotPosition;
        robot.transform.rotation = initialRobotRotation;

        // Reset all ArticulationBodies in the robot to their initial state
        foreach (var state in initialStates)
        {
            // Reset the position and rotation
            state.Item1.transform.localPosition = state.Item2;
            state.Item1.transform.localRotation = state.Item3;

            // Reset the velocity and angular velocity
            state.Item1.velocity = Vector3.zero;
            state.Item1.angularVelocity = Vector3.zero;

            // If the ArticulationBody is not the root, reset the xDrive
            if (state.Item1 != initialStates[0].Item1)
            {
                var drive = state.Item1.xDrive;
                drive.target = 0f;
                state.Item1.xDrive = drive;
            }
        }
    }
```