Unity-Technologies/URDF-Importer #145: Error adding package: https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#v0.5.0

Ro-tico, 2021-09-20:
Hi Unity,

I am following the description in https://github.com/Unity-Technologies/URDF-Importer in order to import a robot in URDF format.  However I get three error in a new project:

[image]

Could you help me please?

sdiao, 2021-09-22:
HI @Ro-tico , may I know which version of Unity that you are using? Do you see the same error if you open a new Unity project, and import the URDF package via `https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#v0.5.0`?

I was reading on [Unity Forum]( https://forum.unity.com/threads/ebusy-resource-busy-or-locked.845497/) where a similar error "EBUSY: resource busy or locked" is reported. Did you have VS Code open when you ran into the error?  If yes, could you try close VS Code first, and re-import `https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#v0.5.0` in Unity Package manager? 

The forum article is: https://forum.unity.com/threads/ebusy-resource-busy-or-locked.845497/

at669, 2021-12-06:
Hi @Ro-tico, as the answer has been provided in this thread, I will close out this issue. Please feel free to re-open the thread if the issue reoccurs. Thank you!

closed 2021-12-06 by at669