Unity-Technologies/URDF-Importer #146: Start supporting file:// type URI

hakuturu583, 2021-09-21:
**Is your feature request related to a problem? Please describe.**
Start supporting file:// type URI.
This change enables us to load mesh files from file:// type URI.
Filename should be any type of URI.
XSD schema is [here](https://github.com/ros/urdfdom/blob/99c6f58120a5dfd1f9fc36419cd9f9aa0246efde/xsd/urdf.xsd#L85).

**Describe the solution you'd like**
Start supporting file:// type URI.

**Describe alternatives you've considered**
None

sdiao, 2021-09-21:
Thanks for submitting the feature request. We are reviewing the requested change.

sdiao, 2021-09-22:
Hi @hakuturu583 , you can also check the [RuntimeUrdfImporterExample Script](https://github.com/Unity-Technologies/URDF-Importer/blob/main/com.unity.robotics.urdf-importer/Runtime/RuntimeImport/RuntimeUrdfImporterExample.cs). You can attach this script  to a Game Object in Unity Editor and use the GUI to load a URDF file. Let us know if it helps. Thanks.

hakuturu583, 2021-09-25:
> Hi @hakuturu583 , you can also check the RuntimeUrdfImporterExample Script. You can attach this script to a Game Object in Unity Editor and use the GUI to load a URDF file. Let us know if it helps. Thanks.

I tried this script and spawn robots like gundam, nao, our team's USV etc...
I will post screenshot to #147

vidurvij-Unity, 2021-10-08:
The pull request was merged and the feature can be accessed in the dev or in the release v0.5.1 which is coming out soon. Thanks for helping us out with this feature.
Vidur

closed 2021-10-08 by vidurvij-Unity