Unity-Technologies/URDF-Importer #161: Friction in URDF ¿?

marcoojer, 2021-12-02:
Hi,
I'm new in Unity and I was trying to increase the friction of the fingers of a gripper to facilitate grasping, but when I load the urdf I can not see in the inspector dialog where to set the friction of just the fingers. Thus, anyone knows how to set it? It would be nice if it could be set in the own urdf.

Thanks in advance

A-Ivan, 2021-12-02:
Hi @marcoojer ,

I haven't tested this out for grasping purposes, but you could look into something called Physic Material, with this you can the adjust static and dynamic friction for individual game objects (in this case the grippers/fingers) by setting this material to that object's collider component.

Here is Unity's [documentation](https://docs.unity3d.com/Manual/class-PhysicMaterial.html) and a [video](https://www.youtube.com/watch?v=SuUNnwswH94&ab_channel=Unity) on how to set this up.

If you don't want to use physics, during the collision of the gripper with the object (grasping) you could [parent](https://docs.unity3d.com/ScriptReference/Transform.SetParent.html) the object to the gripper, this would then have the object move with the gripper (its new parent) until you remove it as the parent later on

I hope this helps. I'm curious if the physic material will help, I assume it will take some time to tinker with and adjust as there are multiple properties that can be set on the physic material.

at669, 2021-12-02:
Hi @marcoojer, the answer above should be what you're looking for--thanks again, @A-Ivan! 

A couple additional comments about getting more accurate physics in your simulation--here are some of the [Physics Settings](https://docs.unity3d.com/Manual/class-PhysicsManager.html) you'll want to look into:
- Changing your `Physics Solver Type` to Temporal Gauss Seidel will reduce erratic behaviors that might be seen with the default solver.
- You may want to change the `Friction Type` to Two Directional Friction Type for applying friction in both tangent directions.
    - In doing so, you'll need to increase the `Default Solver Iterations`, for example to a value like `12`.

Hope this helps!

at669, 2021-12-06:
I'll close out this issue as the answer has been provided, but feel free to open it again if you have further follow-up. Thanks!

closed 2021-12-06 by at669