Unity-Technologies/URDF-Importer #162: Capsule shape support

woven-suomi-tanishige, 2021-12-03:
**Is your feature request related to a problem? Please describe.**
I can't import the URDF with a capsule collider because the current URDF-Importer doesn't support the capsule shape.

**Describe the solution you'd like**
Support capsule shape

at669, 2021-12-03:
Hi, thanks for the submission! I've created an internal ticket to track your PR, and we will update you when it has been reviewed. Thank you!

[Ticket#: AIRO-1625]

vidurvij-Unity, 2022-02-03:
This is an interesting PR and welcome the addition. Just wanted to understand more about the use case. Do you internally us URDFs which involve use of capsules as primitives. As far as the specification goes, its currently not supported by URDF. http://wiki.ros.org/urdf/XML/link

vidurvij-Unity, 2022-02-03:
The branch has been merged

closed 2022-02-03 by vidurvij-Unity