Unity-Technologies/URDF-Importer #165: DllNotFoundException: libvhacd

fanmengcheng, 2021-12-07:
**Describe the bug**
1，in the `URDF Import Settings`, after clicked the button `Import URDF`,it shows  the error `DllNotFoundException: libvhacd`
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '/URDF/niryo_one/'
2. Right Click on 'niryo_one.urdf'
3. Click 'Import URDF'
4. See error

**Console logs / stack traces**
DllNotFoundException: libvhacd
MeshProcess.VHACD.GenerateConvexMeshes (UnityEngine.Mesh mesh) (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/VHACD/VHACD.cs:140)
Unity.Robotics.UrdfImporter.UrdfGeometryCollision.ConvertMeshToColliders (UnityEngine.GameObject gameObject, System.String location, System.Boolean setConvex) (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/Extensions/UrdfGeometryCollision.cs:196)
Unity.Robotics.UrdfImporter.UrdfGeometryCollision.CreateMeshCollider (Unity.Robotics.UrdfImporter.Link+Geometry+Mesh mesh) (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/Extensions/UrdfGeometryCollision.cs:76)
Unity.Robotics.UrdfImporter.UrdfGeometryCollision.Create (UnityEngine.Transform parent, Unity.Robotics.UrdfImporter.GeometryTypes geometryType, Unity.Robotics.UrdfImporter.Link+Geometry geometry) (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/Extensions/UrdfGeometryCollision.cs:44)
Unity.Robotics.UrdfImporter.UrdfCollisionExtensions.Create (UnityEngine.Transform parent, Unity.Robotics.UrdfImporter.Link+Collision collision) (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/Extensions/UrdfCollisionExtensions.cs:62)
Unity.Robotics.UrdfImporter.UrdfCollisionsExtensions.Create (UnityEngine.Transform parent, System.Collections.Generic.List`1[T] collisions) (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/Extensions/UrdfCollisionsExtensions.cs:36)
Unity.Robotics.UrdfImporter.UrdfLinkExtensions.Create (UnityEngine.Transform parent, Unity.Robotics.UrdfImporter.Link link, Unity.Robotics.UrdfImporter.Joint joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/Extensions/UrdfLinkExtensions.cs:27)
Unity.Robotics.UrdfImporter.UrdfRobotExtensions.ProcessJointStack (Unity.Robotics.UrdfImporter.UrdfRobotExtensions+ImportPipelineData im) (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/Extensions/UrdfRobotExtensions.cs:131)
Unity.Robotics.UrdfImporter.UrdfRobotExtensions+<Create>d__8.MoveNext () (at Library/PackageCache/com.unity.robotics.urdf-importer@e9859c4bec/Runtime/Extensions/UrdfRobotExtensions.cs:184)
Unity.EditorCoroutines.Editor.EditorCoroutine+YieldProcessor.MoveNext (System.Collections.IEnumerator enumerator) (at Library/PackageCache/com.unity.editorcoroutines@1.0.0/Editor/EditorCoroutine.cs:80)
Unity.EditorCoroutines.Editor.EditorCoroutine.ProcessIEnumeratorRecursive (System.Collections.IEnumerator enumerator) (at Library/PackageCache/com.unity.editorcoroutines@1.0.0/Editor/EditorCoroutine.cs:134)
Unity.EditorCoroutines.Editor.EditorCoroutine.MoveNext () (at Library/PackageCache/com.unity.editorcoroutines@1.0.0/Editor/EditorCoroutine.cs:115)
UnityEditor.EditorApplication.Internal_CallUpdateFunctions () (at /Users/bokken/buildslave/unity/build/Editor/Mono/EditorApplication.cs:327)

**Screenshots**
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.3.22f1]
- Unity machine OS + version: [e.g. macos High Sierra 10.13.6]

peifeng-unity, 2021-12-07:
Hi @fanmengcheng, the error message indicates that Unity cannot find the libvhacd.dll. Could you double-check if the `libvhacd.dll` is in [the right folder](https://github.com/Unity-Technologies/URDF-Importer/tree/main/com.unity.robotics.urdf-importer/Runtime/VHACD)? If so, please double-check the file's content or the file size to make sure it is not an empty file. Please let us know if the DLL is correctly downloaded or not?

mrpropellers, 2022-01-12:
Closing due to staleness.

closed 2022-01-12 by mrpropellers