Unity-Technologies/URDF-Importer #166: GPU Rendering for Robot Mesh imported from URDF Importer ?

trungle1798, 2021-12-14:
**Is your feature request related to a problem? Please describe.**
Hi Dev Team, I want to ask if there is any solution to implement GPU instancing with the imported mesh from the URDF Importer?
Since I'm training with the imported robot model using Unity ML, I want to render the robot mesh using my GPU. While the training was operating, I realised that my FPS was very low (3-4 FPS) with 25 robot models. 

**Describe the solution you'd like**
Enable GPU Instancing, rendering robot mesh with GPU

**Describe alternatives you've considered**
Reduce the total amount of training environments.

**Additional context**
If there's already a solution that I haven't known, please tell me since my knowledge about GPU rendering is not good.
Thank you guys

peifeng-unity, 2021-12-14:
Hi @trungle1798, thank you for reaching out. I have created an internal ticket (AIRO-1634) to investigate the rendering question. We will come back to you soon.

hyounesy, 2021-12-18:
Hi, @trungle1798, URDF-Importer currently doesn't have the option to enable GPU instancing during the URDF import, but you can try enabling that manually after the URDF is imported: 

- Go to the Materials directorty
- Select all Materials
- In the inspector check the `Enable GPU Instancing`.

[image]

hyounesy, 2022-01-10:
I am closing out this issue. Please feel free to reopen or create a new one if there is any follow-up.

closed 2022-01-10 by hyounesy