Unity-Technologies/URDF-Importer #169: Spherical Joint Tag in URDF supported

WenqiangX, 2022-01-08:
**Is your feature request related to a problem? Please describe.**
Since Unity ArticulationBody has native support of SphericalJoint, as stated in [here](https://docs.unity3d.com/2020.1/Documentation/ScriptReference/ArticulationJointType.html). I wonder why the URDF importer cannot directly support spherical tag in URDF?

I know pybullet community support this feature as mentioned in this [thread](https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=12615).

**Describe the solution you'd like**
I hope I can directly define the spherica joint in URDF with a spherical tag, just like fixed joint, revolute joint, etc.

**Describe alternatives you've considered**
I have two alternative ways to achieve this:
1. A common practice for mimic the spherical joint in URDF is to adopt three revolute joints, however, it requires to add additional virtual objects to link the joints.
2. after import urdf with urdf-importer, manually change the joint type in the ArticulationBody.

**Additional context**
The attachment humanoid.urdf.txt in the pybullet community thread mentioned above, can clearly showcase my point.

Though I have two alternative approach, I actually did not check these two approaches, though I believe the second one is highly doable. Nevertheless, I hope it can be solved natively. Thanks!

at669, 2022-01-10:
Hi, thanks for reaching out! I've created an internal ticket to investigate this, and we'll get back to you with any updates. 

[Ticket#: AIRO-1663]

hijimasa, 2025-07-01:
I introduced spherical joints in the following forked repository.
https://github.com/hijimasa/URDF-Importer
To simplify implementation, I have made the joints free without actuators.
If an actuator is required, please discuss this separately.