Unity-Technologies/URDF-Importer #170: Export robot to urdf: mesh format limits and joint information missing

WenqiangX, 2022-01-13:
**Describe the bug**
A clear and concise description of what the bug is.

I find two bugs (perhaps they are just features) when hitting "export robot to urdf" button.
1. The exporting will try to find stl or dae models. So if I import the urdf with .obj files, it will raise an error that it cannot find the stl or dae model. I think it is quite strange, since exporting is really relevant to the urdf file, I wonder why it must request for the stl or dae format.
2. On each part, if I change the information of Anchor Position, Anchor Rotation and Joint limits (if it is revolute and limited), then I save and export to new urdf. The joint limits will be saved, but anchor position and rotation will not. I assume these three information works together? Since in URDF, there is tag for joint xyz and rpy, I thought it is corresponding to the anchor position and rotation? I wonder why it is not written when exporting? 

**To Reproduce**
I think my description is very clear to reproduce.

**Console logs / stack traces**
Please wrap in [triple backticks (```)](https://help.github.com/en/articles/creating-and-highlighting-code-blocks) to make it easier to read.

for the first bug, the console log is like this:
```
Did not find an existing STL or DAE file for Geometry Mesh f_avg_Pelvis. Exporting a new STL file.
UnityEngine.Debug:Log (object,UnityEngine.Object)
Unity.Robotics.UrdfImporter.UrdfMeshExportHandler:CreateNewStlFile (UnityEngine.GameObject,bool) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/AssetHandlers/UrdfMeshExportHandler.cs:97)
Unity.Robotics.UrdfImporter.UrdfMeshExportHandler:CopyOrCreateMesh (UnityEngine.GameObject,bool) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/AssetHandlers/UrdfMeshExportHandler.cs:52)
Unity.Robotics.UrdfImporter.UrdfGeometry:ExportGeometryMeshData (UnityEngine.GameObject,double[],bool) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfGeometry.cs:305)
Unity.Robotics.UrdfImporter.UrdfGeometry:ExportGeometryData (Unity.Robotics.UrdfImporter.GeometryTypes,UnityEngine.Transform,bool) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfGeometry.cs:41)
Unity.Robotics.UrdfImporter.UrdfCollisionExtensions:ExportCollisionData (Unity.Robotics.UrdfImporter.UrdfCollision) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfCollisionExtensions.cs:71)
Unity.Robotics.UrdfImporter.UrdfCollisionsExtensions/<>c:<ExportCollisionsData>b__1_0 (Unity.Robotics.UrdfImporter.UrdfCollision) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfCollisionsExtensions.cs:45)
System.Linq.Enumerable:ToList<Unity.Robotics.UrdfImporter.Link/Collision> (System.Collections.Generic.IEnumerable`1<Unity.Robotics.UrdfImporter.Link/Collision>)
Unity.Robotics.UrdfImporter.UrdfCollisionsExtensions:ExportCollisionsData (Unity.Robotics.UrdfImporter.UrdfCollisions) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfCollisionsExtensions.cs:45)
Unity.Robotics.UrdfImporter.UrdfLinkExtensions:ExportLinkData (Unity.Robotics.UrdfImporter.UrdfLink) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfLinkExtensions.cs:71)
Unity.Robotics.UrdfImporter.UrdfRobotExtensions:ExportRobotData (Unity.Robotics.UrdfImporter.UrdfRobot) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfRobotExtensions.cs:325)
Unity.Robotics.UrdfImporter.UrdfRobotExtensions:ExportRobotToUrdf (Unity.Robotics.UrdfImporter.UrdfRobot,string,string) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfRobotExtensions.cs:292)
Unity.Robotics.UrdfImporter.Editor.UrdfRobotEditor:OnInspectorGUI () (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Editor/CustomEditors/UrdfRobotEditor.cs:99)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

IndexOutOfRangeException: Index was outside the bounds of the array.
Unity.Robotics.UrdfImporter.StlExporter.TransformMeshToWorldSpace (UnityEngine.Transform meshTransform, UnityEngine.Mesh sharedMesh) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/MeshProcessing/StlExporter.cs:102)
Unity.Robotics.UrdfImporter.StlExporter.CreateWorldSpaceMeshes () (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/MeshProcessing/StlExporter.cs:73)
Unity.Robotics.UrdfImporter.StlExporter.Export () (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/MeshProcessing/StlExporter.cs:35)
Unity.Robotics.UrdfImporter.UrdfMeshExportHandler.CreateNewStlFile (UnityEngine.GameObject geometryObject, System.Boolean isCollisionGeometry) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/AssetHandlers/UrdfMeshExportHandler.cs:103)
Unity.Robotics.UrdfImporter.UrdfMeshExportHandler.CopyOrCreateMesh (UnityEngine.GameObject geometryObject, System.Boolean isCollisionGeometry) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/AssetHandlers/UrdfMeshExportHandler.cs:52)
Unity.Robotics.UrdfImporter.UrdfGeometry.ExportGeometryMeshData (UnityEngine.GameObject geometryObject, System.Double[] urdfSize, System.Boolean isCollisionGeometry) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfGeometry.cs:305)
Unity.Robotics.UrdfImporter.UrdfGeometry.ExportGeometryData (Unity.Robotics.UrdfImporter.GeometryTypes geometryType, UnityEngine.Transform transform, System.Boolean isCollisionGeometry) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfGeometry.cs:41)
Unity.Robotics.UrdfImporter.UrdfCollisionExtensions.ExportCollisionData (Unity.Robotics.UrdfImporter.UrdfCollision urdfCollision) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfCollisionExtensions.cs:71)
Unity.Robotics.UrdfImporter.UrdfCollisionsExtensions+<>c.<ExportCollisionsData>b__1_0 (Unity.Robotics.UrdfImporter.UrdfCollision urdfCollision) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfCollisionsExtensions.cs:45)
System.Linq.Enumerable+SelectArrayIterator`2[TSource,TResult].ToList () (at <351e49e2a5bf4fd6beabb458ce2255f3>:0)
System.Linq.Enumerable.ToList[TSource] (System.Collections.Generic.IEnumerable`1[T] source) (at <351e49e2a5bf4fd6beabb458ce2255f3>:0)
Unity.Robotics.UrdfImporter.UrdfCollisionsExtensions.ExportCollisionsData (Unity.Robotics.UrdfImporter.UrdfCollisions urdfCollisions) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfCollisionsExtensions.cs:45)
Unity.Robotics.UrdfImporter.UrdfLinkExtensions.ExportLinkData (Unity.Robotics.UrdfImporter.UrdfLink urdfLink) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfLinkExtensions.cs:71)
Unity.Robotics.UrdfImporter.UrdfRobotExtensions.ExportRobotData (Unity.Robotics.UrdfImporter.UrdfRobot urdfRobot) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfRobotExtensions.cs:325)
Unity.Robotics.UrdfImporter.UrdfRobotExtensions.ExportRobotToUrdf (Unity.Robotics.UrdfImporter.UrdfRobot urdfRobot, System.String exportRootFolder, System.String exportDestination) (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfRobotExtensions.cs:292)
Unity.Robotics.UrdfImporter.Editor.UrdfRobotEditor.OnInspectorGUI () (at D:/unityPackages/URDF-Importer/com.unity.robotics.urdf-importer/Editor/CustomEditors/UrdfRobotEditor.cs:99)
UnityEditor.UIElements.InspectorElement+<>c__DisplayClass59_0.<CreateIMGUIInspectorFromEditor>b__0 () (at <8df7b5ff67e84cd3a56a03fa707a976e>:0)
UnityEngine.GUIUtility:ProcessEvent(Int32, IntPtr, Boolean&)
```

for the second bug, it will raise no error.

**Expected behavior**
1. Exporting should be consistent with importing, that is, if you can import with obj files, why not exporting with them?
2. The modified joint information should be export to joint tags.

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.15f2c1
- Unity machine OS + version: Windows 10
- ROS machine OS + version: No Ros
- Branch or version: v0.5.0

hyounesy, 2022-01-25:
Thanks for reporting the issues @WenqiangX. We have created internal tickets to further investigate both issues [AIRO-1691] [AIRO-1692]

vidurvij-Unity, 2022-02-06:
It is not a bug, If you refer to the [URDF xml documentation](http://wiki.ros.org/urdf/XML/joint) you will notice that there are two orientations in the joint section.
First is origin:

This sets the orientation of the joint, but if you notice in the image, the successive link's coordinate system becomes same as the joint's coordinate system. This is exactly what URDF importer does. When importing AND EXPORting it reads and write' data in the link' gameobject's transform as done [here](https://github.com/Unity-Technologies/URDF-Importer/blob/9cd43ce496150f9e1351ac478b941c06c099e71c/com.unity.robotics.urdf-importer/Runtime/Extensions/UrdfLinkExtensions.cs#L52).

The second orientation mentioned in the axis section is the one that dictates the anchor rotation and position,  as seen [here](https://github.com/Unity-Technologies/URDF-Importer/blob/9cd43ce496150f9e1351ac478b941c06c099e71c/com.unity.robotics.urdf-importer/Runtime/UrdfComponents/UrdfJoints/UrdfJointContinuous.cs#L133) being set by URDF importer for continuous joint.
and for other joints you can find it in these [files](https://github.com/Unity-Technologies/URDF-Importer/tree/dev/com.unity.robotics.urdf-importer/Runtime/UrdfComponents/UrdfJoints).

So if you want to export the joint data, you must set the link transform and if you want to set the axis data in the URDF you should change the anchor rotation.

vidurvij-Unity, 2022-03-10:
The bug fix has been merged to dev.

closed 2022-03-10 by vidurvij-Unity