Unity-Technologies/URDF-Importer #171: Non-convex MeshCollider with ArticulationBody is not supported

agru97, 2022-01-16:
I imported my Robot withe the newest version of the URDF-Importer, but now I have to disable the All Colliders Convex tag in the URDF Robot Script because when the Colliders are enabled my robot behaves very weird.
When they are disabled I get 153 of these errors:
Non-convex MeshCollider with ArticulationBody is not supported.
If you want to use a non-convex mesh either use the Rigidbody kinematic or remove the ArticulationBody component.

hyounesy, 2022-01-25:
Hi @agru97, thanks for reporting the issue. This is a bit strange, because after importing your robot with URDF-Importer, all the meshes should be decomposed to convex. Have you tried both the `VHACD` and `Unity` for the `Mesh Decomposer` option?

hyounesy, 2022-01-25:
Created an internal ticket to remove the isConvex flag from URDF Robot as it is only supported by RigidBody's which is not supported anymore: [AIRO-1690]

vidurvij-Unity, 2022-02-08:
Changes suggested by the issue are merged.

closed 2022-02-08 by vidurvij-Unity