Unity-Technologies/URDF-Importer #174: BUG: Duplicate DLLs blocking build

abruce33, 2022-01-24:
**Describe the bug**
A clear and concise description of what the bug is.

**When going to build a project for Android (APK), I am getting a build error that is failing the build.  It appears to be duplicate DLLs in both "/x86/assimp.dll "  and "/x86_64/assimp.dll"**

**This is causing the following (4) critical compile errors.**

----
error 1
----
Found plugins with same names, Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll and Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll. Delete the one of the duplicate plugins.
UnityEditor.AndroidPluginImporterExtension:CheckFileCollisions (string)
UnityEditorInternal.PluginsHelper:CheckFileCollisions (UnityEditor.BuildTarget)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

----
error 2
----
Plugins colliding with each other.

----
error 3
----
Build completed with a result of 'Failed' in 14 seconds (14151 ms)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

----
error 4
----
UnityEditor.BuildPlayerWindow+BuildMethodException: 2 errors
  at UnityEditor.BuildPlayerWindow+DefaultBuildMethods.BuildPlayer (UnityEditor.BuildPlayerOptions options) [0x002ca] in <d0e4b8c1204147f3935b6e9c8b8565ad>:0 
  at UnityEditor.BuildPlayerWindow.CallBuildMethods (System.Boolean askForBuildLocation, UnityEditor.BuildOptions defaultBuildOptions) [0x00080] in <d0e4b8c1204147f3935b6e9c8b8565ad>:0 
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

1.)  In my existing Unity 2020.3.17f project, I have "Android" set as the build target.  
2.) I add the URDF-importer package via the Git URL : https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#v0.5.0
3.) After the URDF importer has been added, I click "Build" which starts then stops quickly due to compile errors.
4.) primary error is:
----
Found plugins with same names, Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll and Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll. Delete the one of the duplicate plugins.
UnityEditor.AndroidPluginImporterExtension:CheckFileCollisions (string)
UnityEditorInternal.PluginsHelper:CheckFileCollisions (UnityEditor.BuildTarget)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

**Console logs / stack traces**
Please wrap in [triple backticks (```)](https://help.github.com/en/articles/creating-and-highlighting-code-blocks) to make it easier to read.

----
error 1
----
Found plugins with same names, Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll and Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll. Delete the one of the duplicate plugins.
UnityEditor.AndroidPluginImporterExtension:CheckFileCollisions (string)
UnityEditorInternal.PluginsHelper:CheckFileCollisions (UnityEditor.BuildTarget)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

----
error 2
----
Plugins colliding with each other.

----
error 3
----
Build completed with a result of 'Failed' in 14 seconds (14151 ms)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

----
error 4
----
UnityEditor.BuildPlayerWindow+BuildMethodException: 2 errors
  at UnityEditor.BuildPlayerWindow+DefaultBuildMethods.BuildPlayer (UnityEditor.BuildPlayerOptions options) [0x002ca] in <d0e4b8c1204147f3935b6e9c8b8565ad>:0 
  at UnityEditor.BuildPlayerWindow.CallBuildMethods (System.Boolean askForBuildLocation, UnityEditor.BuildOptions defaultBuildOptions) [0x00080] in <d0e4b8c1204147f3935b6e9c8b8565ad>:0 
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)

**Expected behavior**
A clear and concise description of what you expected to happen.

Clicking "Build" should result in project compiling and exporting an APK.

Troubleshooting note:  The project will build an APK if I remove the URDF-Importer package.

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Environment (please complete the following information, where applicable):**
- Unity Version: [e.g. Unity 2020.2.0f1]
- - - Unity 2020.3.17f
- Unity machine OS + version: [e.g. Windows 10]
- - - Win 10 Razr 14 AMD CPU w/ 3080 NVIDIA GPU
- ROS machine OS + version: [e.g. Ubuntu 18.04, ROS Noetic]
- - - Foxy w/ Noetic ROS 1
- ROS–Unity communication: [e.g. Docker]
- - - TCP-Endpoint is set up but not actively connected. 
- Branch or version: [e.g. v0.2.0]

**Additional context**
Add any other context about the problem here.

I have tried
 - - - Deleting the Library and recompiling -> still fails with duplicate DLLs
 - - - reimporting all assets > still fails with duplicate DLLs

If I remove the URDF-Importer package the project builds normally.

hyounesy, 2022-01-25:
Thanks for reporting the issue @abruce33. Can you please try the following: 
- In the project window, select the AssimpNet.dll file (com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/AssimpNet.dll)  
- In the inspector, exclude it from the android build (https://docs.unity3d.com/Manual/PluginInspector.html)

hyounesy, 2022-01-25:
Internal ticket# [AIRO-1688]

abruce33, 2022-01-25:
Thanks for the direction Hamid, 🙂

It appears the option to exclude AsimpNet.dll from the Android build is disabled? (screenshot attached)

Not sure if this is a bug or a feature, as there does not seem to be any obvious way it has been "locked"

Please advise, thanks!
Adam Bruce

[cid:dea58654-1dac-466c-ba05-68fa21e18426]

abruce33, 2022-01-26:
So the AsimpNet.DLL is managed and properties will not allow me to deselect 'all platforms' so as to exclude Android from the build.  I've started a 'Unity Answers' thread on the DLL unselectable issue here: https://answers.unity.com/questions/1883562/dll-properties-grayed-out-unselectable-unable-to-e.html

abruce33, 2022-01-31:
Update,

> Thanks for reporting the issue @abruce33. Can you please try the following:
>
>     * In the project window, select the AssimpNet.dll file [...]

Update on my end, The Inspector properties for AssimpNet.dll are locked and unselectable.  Therefore I am unable to exclude it from the android build. I am able to access other DLLs in the plungin inspector, but the problematic DLL is locked/unselectable. 

Please let me know if there is an alternative way to deactivate the DLL, or "unlock" it's settings in the plugin Inspector?

Thanks,
Adam

at669, 2022-02-11:
@abruce33, you can open up the package for local editing by adding the package from disk instead of via Git URL--you can follow the steps in the [Workflow](https://github.com/Unity-Technologies/URDF-Importer/blob/main/CONTRIBUTING.md#workflow) section of this doc to unlock these files for editing.

sergiosolorzano, 2022-12-19:
I'm having the same problem in UWP

C:\AR-VR-Github\Robotics-Hub-PandP\Build-Release\Robotics-Hub-PandP\Robotics-Hub-PandP.vcxproj : error  : Cannot load project with duplicated project items: C:\AR-VR-Github\Robotics-Hub-PandP\Build-Release\Robotics-Hub-PandP\assimp.dll is included as 'None' and as 'None' item types. 

@abruce33  Did you solve this?

Thank you, Sergio

VenkataPrasadBoson, 2023-06-14:
delete Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dl and its meta file it works