Unity-Technologies/URDF-Importer #188: Maximum limit of ArticulationBody hierarchy size reached

jbg73, 2022-02-17:
Hi, I'm currently working on my end-of-degree project, which consists of doing some VR tasks with the NAO robot. To do that, the first thing I have to do is to simulate this robot (NAO) in Unity.  The Unity version I'm using is 2020.3.21f1 Personal. 

In order to have the NAO robot in Unity, I have followed these steps:
1. Imported URDF Importer from github (https://github.com/Unity-Technologies/URDF-Importer.git?path=/com.unity.robotics.urdf-importer#v0.5.2)
2. Searched on github for the .urdf files that defines the NAO robot
3. Included a folder in Unity > Project > Assets named nao_description, which contains some subfolders. I go to the path nao_description > urdf > naoV40_generated_urdf and I can find the nao.urdf file and others files such as Materials and nao_meshes.
4. I right-click on the nao.urdf and press Import Robot from Selected URDF file
5. Finally the NAO robot appears in the Scene

Before finishing the import, an error message is shown which says: "The maximum limit of ArticulationBody hierarchy size is reached: hierarchy of articulations can't have more than 64 ArticulationBody nodes!". Besides, during the import process, I can see that this robot has 82 ArticulationBody nodes more or less, so I can understand the origin of this error.

I found another Issue Report on github from someone who has a similar problem but it was from May 2021, and the conclusion was that there was a problem with the Physics Engine and that the Unity team is working on it. My question is if this problem has been solved and maybe my problem is the Unity version that I'm using or if there is another way to solve this. I thought of  dividing the .urdf definition into 2 different files but I don't think this can work.

[image]

Thanks for reading and helping!

A-Ivan, 2022-02-18:
Hi @jbg73 ,

It seems that this issue of 64 ArticulationBody nodes has been solved for Unity versions 2020.3.2f1, 2021.1.0b12, and 2021.2.0a5, [details here](https://issuetracker.unity3d.com/issues/editor-crashes-on-physx-ig-islandsim-activatenode-when-entering-play-mode-and-articulationbody-is-present-in-a-scene). It is strange that you are using Unity version 2020.3.21f1 and are still having this problem.

This was a problem discussed last year [here on github](https://github.com/Unity-Technologies/URDF-Importer/issues/25) and had to do with Nvidia PhysX 4.0 Physics Engine that Unity uses.

I suggest you try to import the URDF using a new Unity 2020.3.2f1 project to just see if the problem still occurs there. Maybe it was fixed for Unity 2020.3.2f1 but then it was broken again in another Unity update.

sarah-gibson, 2022-02-24:
Hi @jbg73, 

The fix that @A-Ivan is referring to corrected the issue where Unity crashed when the ArticulationBody hierarchy is deeper than 64. Now, instead of crashing, you should get a sensible error. But, unfortunately, we still do not support >64 nodes. This is a limitation of Nvidia PhysX 4.1.

Hope this clarifies things!