Unity-Technologies/URDF-Importer #189: How Can I read in Unity the maximun velocity of each joint?

marcoojer, 2022-02-25:
Hi I want to read the maximum velocity of a joint specified in my urdf, in the unity loaded gameobject.

I'm  trying with the value of _articualtionBody.MaxAngularJoint_ and  is  giving me a value of 7 which is the default one, instead of the ones specified in the urdf.

Anyone one knows how to do it?

Thanks in advance!!

vidurvij-Unity, 2022-03-10:
[AIRO-1796]

marcoojer, 2022-10-17:
Thank you for the answer and sorry for delay in the response.
However, I'm loading a urdf with some primatic joint which is like that
`
<joint name="joint_base_finger_2" type="prismatic">
    <origin rpy="0 0 0" xyz="0 0 0"/>
    <parent link="base_tool"/>
    <child link="base_finger_2"/>
    <axis xyz="1 0 0"/>
    <limit effort="140" lower="0" upper="0.025" velocity="0.04166666666666667"/>
</joint>
`

but when I print the ArticulationBody.maxLinearVelocity of the joints it output a value of 100, which I suppose is the default value. Didn't it should be 0.04166