Unity-Technologies/URDF-Importer #192: Importing URDF niryo_one.urdf via Unity 2021.2.16f1 (apple silicon) fails (DllNotFoundException:)

Indie-Siggi, 2022-03-22:
Hello,

I want to report an error which occurred during the tutorial for the Niryo following this tutorial:
Link [Unity-Robotics](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/README.md)

At the tutorial point of importing the Niryo_one.urdf file, an error in the console occurs.
Link [setting-up-the-robot](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/pick_and_place/1_urdf.md#setting-up-the-robot)

I'm working on a MacBook Pro 14" M1Pro with macOS 12.3 using unity 2021.2.16f1 (apple silicon)
ROS is running in Docker container like described in the tutorial

UDFR-Importer 0.5.2

Output Console:

```DllNotFoundException: libvhacd assembly:<unknown assembly> type:<unknown type> member:(null)
MeshProcess.VHACD.GenerateConvexMeshes (UnityEngine.Mesh mesh) (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/VHACD/VHACD.cs:140)
Unity.Robotics.UrdfImporter.UrdfGeometryCollision.ConvertMeshToColliders (UnityEngine.GameObject gameObject, System.String location, System.Boolean setConvex) (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/Extensions/UrdfGeometryCollision.cs:208)
Unity.Robotics.UrdfImporter.UrdfGeometryCollision.CreateMeshCollider (Unity.Robotics.UrdfImporter.Link+Geometry+Mesh mesh) (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/Extensions/UrdfGeometryCollision.cs:80)
Unity.Robotics.UrdfImporter.UrdfGeometryCollision.Create (UnityEngine.Transform parent, Unity.Robotics.UrdfImporter.GeometryTypes geometryType, Unity.Robotics.UrdfImporter.Link+Geometry geometry) (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/Extensions/UrdfGeometryCollision.cs:48)
Unity.Robotics.UrdfImporter.UrdfCollisionExtensions.Create (UnityEngine.Transform parent, Unity.Robotics.UrdfImporter.Link+Collision collision) (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/Extensions/UrdfCollisionExtensions.cs:62)
Unity.Robotics.UrdfImporter.UrdfCollisionsExtensions.Create (UnityEngine.Transform parent, System.Collections.Generic.List`1[T] collisions) (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/Extensions/UrdfCollisionsExtensions.cs:36)
Unity.Robotics.UrdfImporter.UrdfLinkExtensions.Create (UnityEngine.Transform parent, Unity.Robotics.UrdfImporter.Link link, Unity.Robotics.UrdfImporter.Joint joint) (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/Extensions/UrdfLinkExtensions.cs:26)
Unity.Robotics.UrdfImporter.UrdfRobotExtensions.ProcessJointStack (Unity.Robotics.UrdfImporter.UrdfRobotExtensions+ImportPipelineData im) (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/Extensions/UrdfRobotExtensions.cs:132)
Unity.Robotics.UrdfImporter.UrdfRobotExtensions+<Create>d__8.MoveNext () (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/Extensions/UrdfRobotExtensions.cs:186)
Unity.EditorCoroutines.Editor.EditorCoroutine+YieldProcessor.MoveNext (System.Collections.IEnumerator enumerator) (at Library/PackageCache/com.unity.editorcoroutines@1.0.0/Editor/EditorCoroutine.cs:80)
Unity.EditorCoroutines.Editor.EditorCoroutine.ProcessIEnumeratorRecursive (System.Collections.IEnumerator enumerator) (at Library/PackageCache/com.unity.editorcoroutines@1.0.0/Editor/EditorCoroutine.cs:134)
Unity.EditorCoroutines.Editor.EditorCoroutine.MoveNext () (at Library/PackageCache/com.unity.editorcoroutines@1.0.0/Editor/EditorCoroutine.cs:115)
UnityEditor.EditorApplication.Internal_CallUpdateFunctions () (at /Users/bokken/buildslave/unity/build/Editor/Mono/EditorApplication.cs:356)```

sarah-gibson, 2022-03-22:
Hi @Indie-Siggi, Thanks for reporting this issue. We've created an internal ticket to track [AIRO-1834]. Please note that apple silicon is not currently a supported platform, so this ticket will not be prioritized for a while. We'll update you if that changes.

jmachuca77, 2023-04-26:
@sarah-gibson I too am having this issue... is there a way to work around it? We have another machine (windows) where we were able to import our URDF. Is there a way to export the already imported URDF and load it on the Mac?