Unity-Technologies/URDF-Importer #201: URDF Importer causing WebGL build errors in Unity

kyle-rr, 2022-04-28:
I'm using URDF importer tool to import my ROS env into unity. The import is successful. I could also simulate the environment and everything looks to be working fine. However when I try building the unity project for WebGL I get build errors. Looks like these errors are related to URDF Importer. When I remove the URDF importer from the package manager, the build succeeds with no errors, but I may need to use the URDF Importer at runtime later as I add more features to the project. (This bug is essentially the same as the one reported in #122, except my build is targeting WebGL.)

Here's the console output
```Plugin 'assimp.dll' is used from several locations:
 Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll would be copied to <PluginPath>/assimp.dll
 Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll would be copied to <PluginPath>/assimp.dll
Please fix plugin settings and try again.

UnityEditor.Modules.DefaultPluginImporterExtension:CheckFileCollisions (string)
UnityEditorInternal.PluginsHelper:CheckFileCollisions (UnityEditor.BuildTarget)
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)
```
Here's my build options. 
[image]

**Environment:**
- Unity Version: 2021.3.0f1
- Unity machine OS + version: Windows 11
- Branch or version: v0.5.2-preview

patrickyyang, 2023-01-06:
> I'm using URDF importer tool to import my ROS env into unity. The import is successful. I could also simulate the environment and everything looks to be working fine. However when I try building the unity project for WebGL I get build errors. Looks like these errors are related to URDF Importer. When I remove the URDF importer from the package manager, the build succeeds with no errors, but I may need to use the URDF Importer at runtime later as I add more features to the project. (This bug is essentially the same as the one reported in #122, except my build is targeting WebGL.)
> 
> Here's the console output
> 
> ```
>  Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll would be copied to <PluginPath>/assimp.dll
>  Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll would be copied to <PluginPath>/assimp.dll
> Please fix plugin settings and try again.
> 
> UnityEditor.Modules.DefaultPluginImporterExtension:CheckFileCollisions (string)
> UnityEditorInternal.PluginsHelper:CheckFileCollisions (UnityEditor.BuildTarget)
> UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)
> ```
> 
> Here's my build options. [image]
> 
> **Environment:**
> 
> * Unity Version: 2021.3.0f1
> * Unity machine OS + version: Windows 11
> * Branch or version: v0.5.2-preview

Have you fix this error?

Suman7495, 2023-03-09:
Hi, we are running into the same issue. We simply load a basic URDF in Unity and try to build it for WebGL. The build fails with the following errors:
```
Plugin 'assimp.dll' is used from several locations:
 Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll would be copied to <PluginPath>/assimp.dll
 Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll would be copied to <PluginPath>/assimp.dll
Please fix plugin settings and try again.

UnityEditor.BuildPlayerWindow+BuildMethodException: 2 errors
  at UnityEditor.BuildPlayerWindow+DefaultBuildMethods.BuildPlayer (UnityEditor.BuildPlayerOptions options) [0x002cc] in <683db824f2154b7c9f84bd4e86ddfe9f>:0 
  at UnityEditor.BuildPlayerWindow.CallBuildMethods (System.Boolean askForBuildLocation, UnityEditor.BuildOptions defaultBuildOptions) [0x00080] in <683db824f2154b7c9f84bd4e86ddfe9f>:0 
UnityEngine.GUIUtility:ProcessEvent (int,intptr,bool&)
```

**Environment:**

- Unity Version: 2020.3.45f1
- OS: Windows 11
- Branch or version:  v0.5.2-preview

Is there any alternate way to build robot models for WebGL on Unity?

hijimasa, 2025-07-01:
The contents of the above PR have been applied to the following forked repository.
https://github.com/hijimasa/URDF-Importer