Unity-Technologies/URDF-Importer #202: Visual and Collision Mesh Mismatching after re-opening Unity Project

cd25643, 2022-05-18:
Bug Description:
I originally created a URDF using Solidworks with several joints/links, exported both meshes and URDF properties from Solidworks. I then used the URDF importer tool to import the URDF+meshes that I created in Solidworks into Unity. Everything worked as intended. After closing Unity and revisiting a few days later, I now have a weird visual bug/glitch where the visual mesh and the collision mesh do not match. It looks like there are parts missing. I have attatched two images of the same view. The first image is the collision mesh (which still looks correct). The second image is the visual mesh (which should be identical to the collision mesh). Like I mentioned, originally this looked perfectly normal when I first imported the URDF+meshes into Unity. Additionally, I verified the meshes(stl's) in the folder of the URDF itself, and it looks normal. Not sure what caused this, as I did not change anything in the Unity project. The only difference was I closed the project and re-opened it a few days later, and this glitch occurred. Only other piece of info worth noting, the meshes are quite large (file size). The link (base_link in this case) that is having issues is around 120mb. Not sure if that has something to do with the issue...

To Reproduce Bug:
Steps to reproduce the behavior:
1. Import large sized URDF+mesh into Unity (if possible, a URDF created from Solidworks).
2. Verify that the URDF imported into Unity correctly.
3. Save the project, then close Unity.
4. Re-open the project and check to see if the visual meshes are all properly loading in.

Screenshots:
Image_ex1 shows the collision mesh of the URDF link having the issue. As I mentioned above, this is loading in correctly.
Image_ex2 shows the visual mesh of the URDF link having the issue. This should look the same as the collision mesh, and this is the piece that is not rendering in Unity as intended. Even opening the mesh (stl) externally shows a proper mesh, it's only having a issue displaying in Unity.
[image]
[image]

Environment Info:
- Unity Editor Version: 2020.3.27f1
- Unity on Windows 10

matteomeneghetti, 2022-07-04:
I encountered this issue as well. After re-opening my Unity project some visual meshes just disappeared. Re-importing the URDF and deleting the new model works fine as a workaround, but it's a pain for version control.
[image]

- Unity on Windows 11
- Unity Editor Version: 2021.3.5f1

cd25643, 2022-07-06:
> Just for the sake of clarity, for this bug to happen it's sufficient to turn the object into a prefab

@matteomeneghetti  Turning the object into a prefab eliminates this issue? How do you go about doing that?

cd25643, 2022-07-06:
Understood. Only update I have on this issue... is I have found that if mesh files are too large, that is what can cause this issue. I went back and defeatured my URDF meshes extensively, and the issue seems to have gone away...

learning-github-zgy, 2023-11-02:
> No worries @matteomeneghetti. Only update I have on this issue... is I have found that if mesh files are too large, that is what can cause this issue. I went back and defeatured my URDF meshes extensively, and the issue seems to have gone away...

May I ask how you did it? Can you elaborate on how to reduce mesh? Do you operate in the URDF file?