Unity-Technologies/URDF-Importer #212: Cannot include plugin 'Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll' (Key: assimp.dll;None) since plugin with the same name and architecture was already added

QingnanZhong, 2022-11-28:
**Describe the bug**
I got some errors when I export Andorid package.

> Exception: Cannot include plugin 'Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll' (Key: assimp.dll;None) since plugin with the same name and architecture was already added ('Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll (Architecture: None)').
> UnityEditor.Android.AndroidBuildPostprocessor.CollectPlugin (System.String path, Unity.Android.Types.AndroidArchitecture architecture, System.Collections.Generic.Dictionary

What shoud I do?

patricia241, 2023-04-14:
I have the same error.

VenkataPrasadBoson, 2023-06-14:
delete Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll and assimp.dll.meta it works