Unity-Technologies/URDF-Importer #213: UWP Built fresh project doesn't work (VS project unloaded)

sergiosolorzano, 2022-12-19:
**Describe the bug**
A fresh Unity project with imported URDF package builds an sln does throws errors and can't be used to deploy in Hololens 2.

After building in Unity, I open the .sln project but the UWP project says unloaded)
[image]
The only error shown in VS2022 v17.4.3
C:\Temp\testproject\Build-Dev\testproject\testproject.vcxproj : error  : Cannot load project with duplicated project items: C:\Temp\testproject\Build-Dev\testproject\assimp.dll is included as 'None' and as 'None' item types. 

**To Reproduce**
Steps to reproduce the behavior:
1. Create new fresh Unity project Unity 2021.3.15 switch to URP, set .NET Framework api compatibility
2. Add [ROS TCP Connector and URDF importer packages from url](https://github.com/Unity-Technologies/Unity-Robotics-Hub/blob/main/tutorials/quick_setup.md).
3. Build for release, add scene, open visual studio .sln project, the UWP project can't be deployed, VS says project unloaded and VS shows error . VS2022 17.4.3
4. 
[image]

[image]

Only 1 error shown in VS output window:
C:\Temp\testproject\Build-Dev\testproject\testproject.vcxproj : error  : Cannot load project with duplicated project items: C:\Temp\testproject\Build-Dev\testproject\assimp.dll is included as 'None' and as 'None' item types. 

**Console logs / stack traces**
The only error shown in VS is as shown above. Unity does not show any errors when building.

**Expected behavior**
VS sln opens so we can deploy to Hololens 2.

**Screenshots**
If applicable, add screenshots or videos to help explain your problem.

**Environment (please complete the following information, where applicable):**
- Unity Version:  Unity 2021.3.15f1
- Visual Studio 2022 v17.4.3
- Unity machine OS + version: Windows 11 Pro, v 10.0.22.621 Build 22621
- ROS machine OS + version: Didn't launch it because the problem is built project by Unity doesn't work
- ROS–Unity communication: n/a
- Branch or version: 
[image]

**Additional context**
Add any other context about the problem here.

sergiosolorzano, 2023-04-19:
Hey @HVLrobotics unfortunately not.

I tried [as suggested by Unity's Tautvydas-Zilys](https://forum.unity.com/threads/uwp-built-fresh-project-doesnt-work-vs-project-unloaded-and-duplicate-assimp-dll.1376199/#post-8678103) but I couldn't get it to work. 

I think Tautvydas is right: "The real issue is that it seems that this package for whatever reason doesn't have support for all runtime platforms. I don't know why that is, unfortunately. It's either that this DLL was never meant to be used at runtime and was only supposed to be used in the editor (in which case its plugin settings need to be fixed), or the package is still in its infancy and they haven't gotten to making it work on all platforms. We'll have to wait for folks maintaining the package to respond :) (which is likely to not happen until the new year)."

Unity's URDF-Importer didn't respond :(

Sorry I could not be of more help. 
If you figure it out I'd much appreciate you drop a hint here or at the Unity forum thread I posted. 

Good luck, Sergio

bv-radiusmedia, 2024-08-05:
Any update on this guys @sergiosolorzano or @HVLrobotics?

sergiosolorzano, 2024-08-05:
Hi there @bv-radiusmedia no sorry to say, I moved on as I couldnt solve it