Unity-Technologies/URDF-Importer #219: Import Robot from URDF stays greyed out

snowgoon88, 2023-09-26:
**Describe the bug**
I followed the "pick_and_place" tutorial and when trying to import the nyrio_one.urdf file by right-clicking, the "Import Robot from URDF" option stays grey. No possibility to import the URDF.

**Console logs / stack traces**
the only message in the console are warnings
```
Script 'Joint' has the same name as built-in Unity component.
AddComponent and GetComponent will not work with this script.

Assembly for Assembly Definition File 'Assets/RosMessages/Unity.Robotics.RosMessages.asmdef' will not be compiled, because it has no scripts associated with it.
UnityEditor.Scripting.ScriptCompilation.EditorCompilationInterface:CompileScriptsNoStop (UnityEditor.Scripting.ScriptCompilation.EditorScriptCompilationOptions,UnityEditor.BuildTargetGroup,UnityEditor.BuildTarget,string[]) (at /home/bokken/buildslave/unity/build/Editor/Mono/Scripting/ScriptCompilation/EditorCompilationInterface.cs:306)

```

**Screenshots**
file is selected, right click on it
[image]

URDF package seems to be imported in Unity
[image]

**Environment (please complete the following information, where applicable):**
- Unity Version: Unity 2020.3.11f1 Personnal
- Unity machine OS + version: Debian 12 Bookworm
- ROS machine OS + version: Ubuntu 18.04, ROS Melodic
- ROS–Unity communication: Docker
- Branch or version: 
   - for https://github.com/Unity-Technologies/Unity-Robotics-Hub.git
      a72ef3a (HEAD -> main, tag: v0.7.0, origin/main, origin/HEAD) Release 0.7.0

Lando-Ringel, 2025-11-10:
I know this is old, but I was able to solve this problem by moving my Unity PickAndPlaceProject folder to my base C:\ directory. The file path inside the Unity-Robotics-Hub repo is too long for Windows and causes some sort of error. Make sure you delete your Library sub-folder after you move the main project folder.

Noah727, 2025-11-13:
hello, I got the same issue earlier today and it turns out the real niryo_one is the one under niryo_one folder and without the "_urdf". From there, you should be able to import it using URDF.

[image]