Unity-Technologies/URDF-Importer #224: DllNotFoundException: libdl.so, Assimp.Unmanaged

Miftahur92, 2024-01-29:
I am facing problem to use the URDF-importer. The following error only when I only use URDF importer. 

DllNotFoundException: libdl.so assembly:<unknown assembly> type:<unknown type> member:(null)
Assimp.Unmanaged.UnmanagedLibrary+UnmanagedLinuxLibraryImplementation.NativeLoadLibrary (System.String path) (at <d40f2b0ad6814d4c8094142d8473733f>:0)
Assimp.Unmanaged.UnmanagedLibrary+UnmanagedLibraryImplementation.LoadLibrary (System.String path) (at <d40f2b0ad6814d4c8094142d8473733f>:0)
Assimp.Unmanaged.UnmanagedLibrary.LoadLibrary (System.String libPath) (at <d40f2b0ad6814d4c8094142d8473733f>:0)
Assimp.Unmanaged.UnmanagedLibrary.LoadLibrary () (at <d40f2b0ad6814d4c8094142d8473733f>:0)
Assimp.AssimpUnity.InitializePlugin () (at Library/PackageCache/com.unity.robotics.urdf-importer@90f353e435/Runtime/UnityMeshImporter/Plugins/AssimpNet/AssimpUnity.cs:147)

**Environment (please complete the following information, where applicable):**
- Unity 2021.3.25f1
- Unity machine OS - Ubuntu 22.04
- ROS2 Humble
- URDF-importer version - 0.5.2
- ROS-TCP-connector - 0.7.0
- Unity Robotics Visualizations - 0.7.0

**apt-cache search gave me following results:**
assimp-testmodels - assimp 3D model import library (testdata)
assimp-utils - 3D model import library (utilities)
gem-plugin-assimp - Graphics Environment for Multimedia - Assimp support
libapache2-authcassimple-perl - Apache2 module to authenticate trough a CAS server
libassimp-dev - 3D model import library (development)
libassimp-doc - 3D model import library (documentation)
libassimp5 - 3D model import library
python3-pyassimp - 3D model import library (Python3 bindings)
qt3d-assimpsceneimport-plugin - Qt 3D GL Assimp scene import plugin
qt6-3d-assimpsceneimport-plugin - Qt 6 3D GL Assimp Scene Import plugin
qt6-quick3d-assetimporters-plugin - Qt 6 Quick 3D Assimp Import plugin
ros-humble-rviz-assimp-vendor - Wrapper around assimp, providing nothing but a dependency on assimp, on some systems.
ros-iron-rviz-assimp-vendor - Wrapper around assimp, providing nothing but a dependency on assimp, on some systems.
ros-rolling-rviz-assimp-vendor - Wrapper around assimp, providing nothing but a dependency on assimp, on some systems.

WadeNotCode, 2024-03-17:
Bro, have u ever found the solution?

Miftahur92, 2024-03-19:
> Bro, have u ever found the solution?

Nope. I think the problem is with Ubuntu. I am using the same setup with WSL2, and it is working fine.

Cryoscopic-E, 2024-03-23:
Problem is with Assimp library not using libdl.so.2 that is now in newest Debian, Ubuntu, etc. I got tired of the problem and fixed with symbolic linking libdl.so.2 as `sudo ln -s /usr/lib/libdl.so.2 /usr/lib/libdl.so`. Although is no recommended messing with /usr/lib but really can't be bothered to wait for a fix.