Unity-Technologies/URDF-Importer #228: namespace 'UrdfImporter' does not exist in 'Unity.Robotics'

FabianEP11, 2024-09-25:
I have a Unity project with the URDF-Importer plugin (v0.5.2), and I need to get the name of the joints from code like this: 

`string joint_name = articulationBody.gameObject.GetComponent<UrdfJointRevolute>().jointName;`
which uses `using Unity.Robotics.UrdfImporter `.

When I run the app in the Editor everything works fine, but when I build the app, it fails because it cannot fine the namespace:

`The type or namespace name 'UrdfImporter' does not exist in the namespace 'Unity.Robotics' (are you missing an assembly reference?)`

I also tried adding the entire namespace to the GetComponent type like:  `string joint_name = articulationBody.gameObject.GetComponent<Unity.Robotics.UrdfImporter.UrdfJointRevolute>().jointName;` but it is the same 

Is the `using Unity.Robotics.UrdfImporter` just available on the Editor? or how can I fix that?

CarlTornberg, 2024-11-18:
The URDF importer needs to compile for whichever build target you have set (In my case, I use Android).

First, move the URDF importer package from the CachedPackage folder to the asset package folder:
1. Find the package by: Project -> Packages -> URDF Importer
2. Right-click and press Show in explorer,  ctrl+x the URDF folder, 
3. Paste in the Asset/Packages by:  Project -> Assets 
4. Right-click and press Show in explorer.
5. Paste in the Packages folder 

When the package is moved, let it recompile and find the package in the editor.
Project -> Packages -> URDF Importer -> Runtime -> Unity.Robotics.URDFImporter -> (In the inspector) Click with platform you are compiling to -> Apply.

[image]

FabianEP11, 2024-11-18:
Hey,
Yeah, previously I was able to build the app for a HoloLens device by manually adding "WSA" to the "includePlatforms" in the "Library\PackageCache\com.unity.robotics.urdf-importer@90f353e435\Runtime\Unity.Robotics.URDFImporter.asmdef" file. 
I guess that the inspector checkbox changes the same file. 

Thanks!

CarlTornberg, 2024-11-19:
Further more, Im not sure if this is for all build platforms, but is for Android. 
An exception named `'Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86/assimp.dll' (Key: assimp.dll;None) since plugin with the same name and architecture was already added ('Packages/com.unity.robotics.urdf-importer/Runtime/UnityMeshImporter/Plugins/AssimpNet/Native/win/x86_64/assimp.dll (Architecture: None)')` fails the build.
[Here is the solution to that issue](https://github.com/Unity-Technologies/URDF-Importer/issues/212#issuecomment-1591807545)

lidafei123, 2025-01-02:
Hey,
I have the same problem can't find namespace:The type or namespace name 'UrdfImporter' does not exist in the namespace 'Unity.Robotics' (are you missing an assembly reference?)
I followed your steps and still can't solve it. how can I fix that?[image]

frank-Regal, 2025-02-04:
Editing the `Unity.Robotics.URDFImporter.asmdef` file worked for me! Here is what my modified file looks like.

{
    "name": "Unity.Robotics.UrdfImporter",
    "rootNamespace": "Unity.Robotics.UrdfImporter.Urdf",
    "references": [
        "GUID:6a697808d7c80a549b57420070d6c4f3"
    ],
    "includePlatforms": [
        "Editor",
        "LinuxStandalone64",
        "macOSStandalone",
        "WindowsStandalone64",
        "WSA"
    ],
    "excludePlatforms": [],
    "allowUnsafeCode": true,
    "overrideReferences": false,
    "precompiledReferences": [],
    "autoReferenced": true,
    "defineConstraints": [],
    "versionDefines": [],
    "noEngineReferences": false
}

The file is located in `<your_unity_project>\Library\PackageCache\com.unity.robotics.urdf-importer@90f353e435\Runtime` as noted above.